margelo/react-native-vision-camera #628: ✨ Need a webrtc frame processing plug-in.

mushan0x0, 2021-12-06:
### What feature or enhancement are you suggesting?

Cooperate with react-native-vision-camera to make videos.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

not yet.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2021-12-06:
Sure, let me know when you build one and I can add it to the "Community Plugins" section in the docs :)

closed 2021-12-06 by mrousavy

umarshaikh64, 2022-01-05:
hello tell me how  can we app webrtc in  react-native-camera-vision thanks

mrousavy, 2022-01-05:
Read the Frame Processors docs.

salmanrajz, 2022-01-10:
> Read the Frame Processors docs.

Hey, I didn't find any solution, regarding WebRTC implementation, could you share the example if possible, it may help to lots of developers.

fukemy, 2023-06-07:
@salmanrajz did u found any solution

salmanrajz, 2023-06-11:
> @salmanrajz did u found any solution

Not yet

fukemy, 2023-06-15:
@salmanrajz I am trying to integrate with WebRTC, here is the way I am trying to: 

1. Create custom capture as global static class
2. Add it's into WebRTC videoCapturer
3. Inside frame processor, access to capture delete with new Frame

```
if let stream = webrtcModule.stream(forReactTag: localStreamTag){
      if let videoTrack = stream.videoTracks.first{
        if let videoCapturerController = videoTrack.captureController as? VideoCaptureController{
          videoCapturerController.capturer.stopCapture{
            let visionCapturer = VisionCameraCustomCapturer(delegate: videoTrack.source)
            let newVideoCapturer = RTCCameraVideoCapturer(delegate: visionCapturer.delegate!)
            let newVideoCapturerController = VideoCaptureController(capturer: newVideoCapturer, andConstraints: nil)
            videoTrack.captureController = newVideoCapturerController
            newVideoCapturerController?.startCapture()
            PipInstance.shared().visionCapturer = visionCapturer
          }
        }
      }
    }
```

```
@objc(VisionFrameProcessorPlugin)
public class VisionFrameProcessorPlugin: NSObject, FrameProcessorPluginBase {
  @objc public static func callback(_ frame: Frame!, withArgs _: [Any]!) -> Any! {
    guard let imageBuffer = CMSampleBufferGetImageBuffer(frame.buffer) else {
      return nil
    }
    let filter = YUCIHighPassSkinSmoothing()
    filter.inputImage = CIImage(cvImageBuffer: imageBuffer)
    filter.inputAmount = 0.8
    if let buffer = filter.outputImage?.pixelBuffer{
      if let newCmSampleBuffer = buffer.toCMSampleBuffer1(){
        print("newCMSamplBuffer: \(newCmSampleBuffer)")
        if #available(iOS 15.0, *) {
          if let visionCapturer = PipInstance.shared().visionCapturer{
            print("visionCapturer: \(visionCapturer)")
            let rtcPixelBuffer = RTCCVPixelBuffer(pixelBuffer: buffer)
            let frameTime = CMSampleBufferGetOutputPresentationTimeStamp(frame.buffer);
            let frameTimeMillisecs = Int64(CMTimeGetSeconds(frameTime) * 1000);
            var rotation = RTCVideoRotation(rawValue: 0)
            switch(frame.orientation){
            case .left:
              rotation = RTCVideoRotation(rawValue: 90)
            case .down:
              rotation = RTCVideoRotation(rawValue: 180)
            case .right:
              rotation = RTCVideoRotation(rawValue: 270)
            default:
              rotation = RTCVideoRotation(rawValue: 0)
            }
            let newVideoFrame = RTCVideoFrame(buffer: rtcPixelBuffer, rotation: rotation!, timeStampNs: frameTimeMillisecs)
            print("newVideoFrame: \(newVideoFrame)")
            visionCapturer.delegate?.capturer(visionCapturer, didCapture: newVideoFrame)
          }
        }
        return Frame(buffer: newCmSampleBuffer, orientation: frame.orientation)
      }
    }else if let buffer = filter.outputImage?.toCVPixelBuffer(){
      if let newCmSampleBuffer = buffer.toCMSampleBuffer1(){
        if #available(iOS 15.0, *) {
          if let visionCapturer = PipInstance.shared().visionCapturer{
            print("visionCapturer: \(visionCapturer.delegate)")
            let rtcPixelBuffer = RTCCVPixelBuffer(pixelBuffer: buffer)
            let frameTime = CMSampleBufferGetOutputPresentationTimeStamp(frame.buffer);
            let frameTimeMillisecs = Int64(CMTimeGetSeconds(frameTime) * 1000);
            var rotation = RTCVideoRotation(rawValue: 0)
            switch(frame.orientation){
            case .left:
              rotation = RTCVideoRotation(rawValue: 90)
            case .down:
              rotation = RTCVideoRotation(rawValue: 180)
            case .right:
              rotation = RTCVideoRotation(rawValue: 270)
            default:
              rotation = RTCVideoRotation(rawValue: 0)
            }
            let newVideoFrame = RTCVideoFrame(buffer: rtcPixelBuffer, rotation: rotation!, timeStampNs: frameTimeMillisecs)
            visionCapturer.delegate?.capturer(visionCapturer, didCapture: newVideoFrame)
          }
        }
        return Frame(buffer: newCmSampleBuffer, orientation: frame.orientation)
      }
    }
    return frame
  }
}
```

I think it's will work, could you make it work with me?