margelo/react-native-vision-camera #633: ❓ Enable Flash when using frameProcessor

jensdev, 2021-12-07:
### Question

How can I enable the flash light while using a frameProcessor. I'm using the Colorwaver example at the moment. Also thank you for this amazing package

### VisionCamera Version

2.9.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-12-07:
Hey! Thanks for the kind words, you can pass `torch={true}`. :)

closed 2021-12-07 by mrousavy

jensdev, 2021-12-07:
Thanks for your swift reply. `torch={'on'}` did the trick