margelo/react-native-vision-camera #634: 🐛 CameraViewManager cannot be cast to ViewGroupManager

willcray, 2021-12-08:
### What were you trying to do?

I'm trying to run the app and use the camera screen on an Android device.

### What happened instead?

I'm receiving this error when I open up the camera screen of my app. Everything builds fine. The app runs with no warnings. 

Then, I receive `com.mrousavy.camera.CameraViewManager cannot be cast to com.facebook.react.uimanager.ViewGroupManager` when I open the camera screen.

react-native: 0.66.3
rnvc: 2.9.4

It appears this issue has been encountered before. Did we find a solution?:

Okay so first off those are all very different errors.

---

> the error CameraViewManager cannot be cast to ViewGroupManager happened

This seems really weird. I have never seen that error before and do not know how to reproduce it, if you can provide me a GitHub repository that I can run to reproduce it that would help me.

---

> but when refreshing the c++ linking this happens
>  `ninja: error: '../../../../build/react-native-0.64.2.aar/jni/armeabi-v7a/libjscexecutor.so', needed by '../../../../build/intermediates/cmake/debug/obj/armeabi-v7a/libVisionCamera.so', missing and no known rule to make it`

This should be fixed by running the build twice. See https://github.com/mrousavy/react-native-vision-camera/issues/338

---

> Now I did also try running the example project. Downloaded it and ran yarn install. Try running it on the phone with "Run 'app'" and got error:
> `ninja: error: "../../../../build/third-party-ndk/hermes/jni/arm64-v8a/libhermes.so", needed by "../../../../build/intermediates/cmake/debug/obj/arm64-v8a/libVisionCamera.so", missing and no known rule to make it`

Same as above. See https://github.com/mrousavy/react-native-vision-camera/issues/338. Also you want to run `yarn` in both the root directory and in the `example/` directory. Then open `example/android/` in Android Studio.

---

> But another error "parallel-video-processing-not-supported".

This happens in the Example app? See ["the `supportsParallelVideoProcessing` prop"](https://mrousavy.github.io/react-native-vision-camera/docs/guides/devices#the-supportsparallelvideoprocessing-prop), or related issue: https://github.com/mrousavy/react-native-vision-camera/issues/315

---

Maybe this whole thing has to do with you using an M1 laptop? the `CameraViewManager cannot be cast to ViewGroupManager` still seems so weird to me....

_Originally posted by @mrousavy in https://github.com/mrousavy/react-native-vision-camera/issues/388#issuecomment-904680871_

### Device

Motorola Moto E (physical device), no issue on iPhone 11 (physical device)

### VisionCamera Version

2.9.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

willcray, 2021-12-10:
@mrousavy @trta2352 did you all ever find a solution to #388?

willcray, 2021-12-10:
My bad! Issue derived from using SafeAreaView in combination with the Camera component.

closed 2021-12-10 by willcray

didithilmy, 2022-01-06:
@willcray can you please elaborate what the solution was? I encountered the same problem but I couldn't find the solution. Much appreciated 😄

willcray, 2022-01-12:
> @willcray can you please elaborate what the solution was? I encountered the same problem but I couldn't find the solution. Much appreciated 😄

Yes, it was a simple fix. I stopped using `SafeAreaView` in the same JSX component as the `Camera`. I removed it from the component that renders the camera.

arthwood, 2022-01-31:
I have exactly the same error and I may know what is causing the issue. In my case I needed to render some elements on top of camera view so I'm rendering the as a children of `Camera` element. It works fine on iOS but crashes with:

> com.mrousavy.camera.CameraViewManager cannot be cast to com.facebook.react.uimanager.ViewGroupManager

I'm not using `SafeAreaView` as `Camera` parent but regular `View`.

MrMole96, 2022-02-17:
Yes, i am facing the same issue with Android. Did you guys find work around?

vladimirevstratov, 2022-03-13:
Same issue, dont have SafeAreaView but just View.

gary-cohen, 2022-03-15:
I'm also seeing the same issue.