margelo/react-native-vision-camera #635: Testflight build is crashing on starting camera and recording Audio.🐛

Tahir-embrace, 2021-12-09:
### What were you trying to do?

I am trying to start a video from front camera and recording the audio there.

### What happened instead?

The starting video and recording audio is working fine on the DEV,
when i try this on Testflight app crashed.

here are the crash logs i got from Firebase.

### Relevant log output

```shell
Crashed: mrousavy/VisionCamera.audio
0  AppName                         0x254dc0 specialized CameraView.captureOutput(_:didOutput:from:) + 1119100 (CameraView+RecordVideo.swift:1119100)
1  AppName                         0x254680 @objc CameraView.captureOutput(_:didOutput:from:) + 1117244 (<compiler-generated>:1117244)
2  AVFCapture                     0x70bd4 -[AVCaptureAudioDataOutput _handleSampleBufferEventForSampleBuffer:] + 224
3  AVFCapture                     0x70960 __47-[AVCaptureAudioDataOutput _updateRemoteQueue:]_block_invoke + 88
4  CMCapture                      0x2303c __FigRemoteOperationReceiverCreateMessageReceiver_block_invoke + 276
5  CMCapture                      0x2874c4 __rqReceiverSetSource_block_invoke + 248
6  libdispatch.dylib              0x61298 _dispatch_client_callout + 16
7  libdispatch.dylib              0x39a70 _dispatch_continuation_pop$VARIANT$armv81 + 404
8  libdispatch.dylib              0x4a31c _dispatch_source_invoke$VARIANT$armv81 + 1248
9  libdispatch.dylib              0x3d210 _dispatch_lane_serial_drain$VARIANT$armv81 + 260
10 libdispatch.dylib              0x3de2c _dispatch_lane_invoke$VARIANT$armv81 + 400
11 libdispatch.dylib              0x4766c _dispatch_workloop_worker_thread + 692
12 libsystem_pthread.dylib        0xb5bc _pthread_wqthread + 272
13 libsystem_pthread.dylib        0xe86c start_wqthread + 8
```

### Device

iPhone 8: iOS 14.7.1, iPhone 7: iOS 15

### VisionCamera Version

2.9.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-12-09:
hmm 🤔

Tahir-embrace, 2021-12-09:
@mrousavy updated the issue description 
please review it might be little more helpful.

mrousavy, 2021-12-09:
Does it tell you the exact line in that file? Which Swift version are you using? Are you on `SWIFT_VERSION=5.2`?

Tahir-embrace, 2021-12-09:
@mrousavy not any specific line ... it just lands on this file..

and swift version is: 
`swift-driver version: 1.26.9
Apple Swift version 5.5.1 (swiftlang-1300.0.31.4 clang-1300.0.29.6)
Target: x86_64-apple-macosx11.0
`

tahir-zahid, 2022-01-26:
@mrousavy it lands on file CameraView+RecordVideo.swift and line number 196
[image]

mrousavy, 2022-01-26:
Why can `masterClock` be `nil`??? 🤨

tahir-zahid, 2022-01-26:
@mrousavy thanks for quick response ...
Actually i am not a iOS native person but struggling a lot for it.

mrousavy, 2023-09-30:
This is now fixed with V3! 🥳

closed 2023-09-30 by mrousavy