margelo/react-native-vision-camera #638: 🐛 Links on docs/api main page are 404

pke, 2021-12-09:
### What were you trying to do?

Links on main page https://mrousavy.com/react-native-vision-camera/docs/api are all 404

[image]

### What happened instead?

All links on main page are 404.

### Device

Desktop

### VisionCamera Version

n/a

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-12-09:
Ah that sucks. Do you by any chance have experience with docusaurus + typedoc?

pke, 2021-12-09:
Unfortunately I am not familiar with it. Probably just a config issue. Maybe it could help to file a bug over at docusaurus and point to this repo?

mrousavy, 2023-09-30:
Fixed I think!

closed 2023-09-30 by mrousavy