margelo/react-native-vision-camera #653: 🐛 Not using frame processors results in zoomed image

savv, 2021-12-14:
### What were you trying to do?

I wanted to take a picture, using the app under [example/](https://github.com/mrousavy/react-native-vision-camera/tree/1f4d188/example).

CameraPage initially works as expected; however, commenting out https://github.com/mrousavy/react-native-vision-camera/blob/1f4d188c5267f443867f6aae1548fb27e32d9931/example/src/CameraPage.tsx#L232 results in the image being zoomed out.

With frame processors:

Without frame processors:

### Reproduceable Code

```tsx
Comment out https://github.com/mrousavy/react-native-vision-camera/blob/1f4d188c5267f443867f6aae1548fb27e32d9931/example/src/CameraPage.tsx#L232 and run on an SM-T500 tablet (samsung galaxy tab a7).
```

### What happened instead?

With frame processors:
[image]

Without frame processors (no zooming on my part):
[image]

### Device

SM-T500

### VisionCamera Version

v2.10.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

savv, 2021-12-14:
This seems to be a bug with the logic with how the format is selected in the example.

closed 2021-12-14 by savv

Alex1899, 2022-02-24:
has this been fixed? Having this issue on android