margelo/react-native-vision-camera #654: Build keep failing on task ':react-native-vision-camera:extractJNIFiles'

robsonbbs, 2021-12-15:
### What were you trying to do?

Build my app after installing react-native-vision-camera

FYI:  I am using Expo SDK 41 with RN 0.63.4.

### What happened instead?

Build fail on task ':react-native-vision-camera:extractJNIFiles'

### Relevant log output

```shell
[stderr] FAILURE: Build failed with an exception.
[stderr] * Where:
[stderr] Build file '/root/workingdir/build/node_modules/react-native-vision-camera/android/build.gradle' line: 455
[stderr] * What went wrong:
[stderr] Execution failed for task ':react-native-vision-camera:extractJNIFiles'.
[stderr] > Cannot expand ZIP '/root/workingdir/build/node_modules/react-native-reanimated/android/react-native-reanimated-63-jsc.aar' as it does not exist.
[stderr] * Try:
[stderr] Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
[stderr] * Get more help at https://help.gradle.org
[stderr] BUILD FAILED in 12m 52s
Error: Gradle build failed with unknown error. Please see logs for the "Run gradlew" phase.
```

### Device

No device. This problem happening while running eas build.

### VisionCamera Version

2.8.3

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

robsonbbs, 2021-12-15:
One strange thing that I noticed: 

The error output claims that `react-native-reanimated-63-jsc.aar`  wasn't found, and it's true, I don't have this file on my `/node-modules/react-native-reanimated/android`, but I have `react-native-reanimated-63.aar`. So the point is: Why this task is looking for `...63-jsc.aar` since the file name doesn't have this JSC?

I am starting to suspect that I am missing something, and there is some specific version to be used with Expo SDK 41 and RN 0.63. 

Could anyone please help me figure this out?

viljark, 2021-12-16:
Expo support progress can be seen here https://github.com/mrousavy/react-native-vision-camera/pull/545. Combining this PR's branch with [mentioned fix](https://github.com/mrousavy/react-native-vision-camera/pull/545#issuecomment-977187512) in master branch results in sucessful build.

closed 2022-01-02 by mrousavy