margelo/react-native-vision-camera #659: Using Frame Processor outside RN component✨

jared-94, 2021-12-16:
### What feature or enhancement are you suggesting?

Hi,

I would like to use your package, in particular Frame Processor in a **foreground service** (Android only). Thus even when the app is in a killed state.

But it seems that using this implies to return a JSX component `<Camera />`, or maybe I miss something...

Is this use case possible?

Thanks

### What Platforms whould this feature/enhancement affect?

Android

### Alternatives/Workarounds

No workarround found...

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2021-12-16:
Hey Jared!

Can you give me an example repo/code? I've never had such a use case before

jared-94, 2021-12-17:
Thank you for your answer !

Unfortunately, I do not have code or repo. 
But I'm using rn-foregroud-service 
https://github.com/Raja0sama/rn-foreground-service

It allows to run JS task in the foreground service.
Thus, I need a pure JS function, which of course may call native module, to process Frame Processor.
In this kind of function, I can't use any hook, or return JSX component.

Do you think your lib may be adjusted for such an use case?

suniahk, 2021-12-17:
An example of what you're trying to achieve would helpful, as right now it's kinda hard to understand the use case you're talking about.  From the sound of it, you're trying to process frames from a camera component that isn't rendered (meaning there are no frames coming from it to process in the first place).  It sounds like you'd be better served by capturing the frame/photo, storing it to a state object, and processing it from that instead of trying to hook into the frame processor. 

Have you looked at React Native's in-library NativeEmitter class?  That might be of more use to you for this.  Like I said though, an example of your specific use case would help a lot.

jared-94, 2021-12-18:
OK I see.

What I want to achieve in this part of my app, is to capture motion detection from front camera, and then make an http request to a home automation software, so that the device acts as a PIR motion sensor.
But I need this in a foreground service, without rendering anything, and without user interaction.

So maybe I need a loop that compare the frame at time t-1 with the one at time t and a kind of AI that may detect a change (for exemple if light is turn on/off in the room, the picture will change, but there is no motion...)

Do you have any idea how I can achieve that ?

Thanks for your help!

Sid-Turner-Ellis, 2023-02-22:
@jared-94 did you manage to get this going?

Facing similar problems - https://github.com/mrousavy/react-native-vision-camera/issues/1011

mrousavy, 2023-09-30:
Hey, it needs to render in an app.

closed 2023-09-30 by mrousavy