margelo/react-native-vision-camera #662: 🐛Custom FrameProcessor plugin crashes the android app when pushing very long string into WritableNativeArray (or WritableNativeMap)

FloweryK, 2021-12-18:
### What were you trying to do?

Hi, I'm making a custom frame processor that returns very long string (such as base64), so that i can send each frame's data to my personal server.

### Reproduceable Code

```tsx
public class SampleFrameProcessorPlugin extends FrameProcessorPlugin {
    @Nullable
    @Override
    public Object callback(@NonNull ImageProxy image, @NonNull Object[] params) {
        WritableNativeArray array = new WritableNativeArray();
        String testString = new String(new char[200000]).replace("\0", "a");

        array.pushString(testString); // I found that this line Crashes the app.

        return array;
    }

    SampleFrameProcessorPlugin() {
        super("sampleProcessor");
    }
}
```

### What happened instead?

But I found that WritableNativeArray (or WritableNativeMap) crashes my app without giving any error, after certain amount time.

Here's what I've tested so far

1. Tried putting a single string with length 200,000 to WritableNativeArray (like above code), and the app crashes after processing like 2 frames.
2. Tried putting 1,000 strings with length 200  (so that the array can contain the same data with split manner), and the app crashes just as above (crashed after within few frames)
3. Tried putting 200 strings with length 2, and the app crashes after 2~3 minutes.
4. Tried putting a single string with length 200 to WritableNative Array, the app looks fine. But not sure it would be ok forever or not.

At first, I was not sure whether this is React Bridge issue or VisionCamera issue. Then I made a custom native bridge module (without using VisionCamera) that pushes string with length 200,000 just I tested above, then the app was fine. So this might be an issue with VisionCamera.

My best guess is that FrameProcessor Plugin saves certain amount of previous data, which soon exceeds the memory limit of the android app and kills the app. 

Any idea about this error or how to fix this?

thank you.

### Relevant log output

```shell
No error comes out. The app just dies.
```

### Device

Galaxy Note10+ (Real device), Pixel 2 API 29 (AVD)

### VisionCamera Version

2.10.0 (latest, installed with expo)

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-12-18:
This seems like a wrong access after a value has been garbage collected by the JS runtime, so somewhere I am probably dereferencing an already deleted value in C++ - causing it to crash. I currently don't have a lot of time to investigate this, but I hope to have that time soon! And if not, I surely can help you investigate this by giving you some pointers if you want to look at this yourself :)

btw why are you sending each frame to your server? lol

mrousavy, 2021-12-18:
Also, adb logs can be helpful

FloweryK, 2021-12-19:
@mrousavy well, I'm making a machine learning app that processes camera preview (like instagram or tiktok's filtering). I want to process each frame with my model, and soon found out that my model would be too big and too slow for mobile phones. So I decided to send each frame from android to my server, process it, and send back to android. My first shot was to send each frame as base64. That's why im trying to translate into base64, and send it to my server.

Right now I'm transforming native Imageproxy into string(base64) on native code... which soon turned out to kill my app ;( 
If native bridge can send Bitmap format directly to react native, it would be also fine so that i can transform ImageProxy into Bitmap on native code, send Bitmap to react native, and transform it into base64 on there. But i haven't found a way to DIRECTLY send bitmap format (or any other format available rather than WritableNativeMap or WritableNativeArray) 😢 

Anyway, i'll take a look at adb log as soon as I get back home!

FloweryK, 2021-12-28:
I found a similar issue #587. So I'll just work on that issue and close this one.

closed 2021-12-28 by FloweryK