margelo/react-native-vision-camera #664: 🐛  Android crashes when resuming to app after host activity is destroyed

priittiidla, 2021-12-20:
### What were you trying to do?

Hi. I am using this library to take pictures in an application I am developing.
I found that app crashes on camera screen when user wants to return to app from background. 

### What happened instead?

Looking into issue more deeply I found that crash happens when host activity had been destroyed while app was in background. (Using "Don't keep activities" to simulate this)

Removing lifecycle event listener when LifecycleEventListener.onHostDestroy() method is called resolves the issue for me.

### Relevant log output

```shell
Caused by java.lang.IllegalStateException: no event up from DESTROYED
       at androidx.lifecycle.LifecycleRegistry.forwardPass(LifecycleRegistry.java:263)
       at androidx.lifecycle.LifecycleRegistry.sync(LifecycleRegistry.java:307)
       at androidx.lifecycle.LifecycleRegistry.moveToState(LifecycleRegistry.java:148)
       at androidx.lifecycle.LifecycleRegistry.setCurrentState(LifecycleRegistry.java:121)
       at com.mrousavy.camera.CameraView.updateLifecycleState(CameraView.kt:258)
       at com.mrousavy.camera.CameraView.access$updateLifecycleState(CameraView.kt:65)
       at com.mrousavy.camera.CameraView$3.onHostResume(CameraView.kt:213)
       at com.facebook.react.bridge.ReactContext.onHostResume(ReactContext.java:263)
       at com.facebook.react.ReactInstanceManager.moveToResumedLifecycleState(ReactInstanceManager.java:741)
       at com.facebook.react.ReactInstanceManager.onHostResume(ReactInstanceManager.java:640)
       at com.facebook.react.ReactInstanceManager.onHostResume(ReactInstanceManager.java:590)
       at com.facebook.react.ReactDelegate.onHostResume(ReactDelegate.java:53)
       at com.facebook.react.ReactActivityDelegate.onResume(ReactActivityDelegate.java:100)
       at com.facebook.react.ReactActivity.onResume(ReactActivity.java:58)
       at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1286)
       at android.app.Activity.performResume(Activity.java:6987)
       at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4144)
       at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:4245)
       at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3360)
       at android.app.ActivityThread.access$1100(ActivityThread.java:221)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1794)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:158)
       at android.app.ActivityThread.main(ActivityThread.java:7224)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
```

### Device

Samsung Galaxy S8 (Android 9), Samsung Galaxy S10 (Android 11)

### VisionCamera Version

2.10.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-12-20:
Interesting, maybe the CameraX upgrade helps here

priittiidla, 2021-12-21:
I believe it's not CameraX issue. 
It's connected to not clearing/releasing used objects when lifecycle is destroyed.

mrousavy, 2022-01-02:
Do you have a fix in mind?

priittiidla, 2022-01-03:
https://github.com/priittiidla/react-native-vision-camera/pull/1
Removing lifecycle event listener fixes issue for my use case.

closed 2022-01-03 by mrousavy

faustaleonardo, 2022-06-24:
Thank you @priittiidla

firatbadur, 2023-08-28:
> 

Hi, your solution link is not working. Can you share the details of solving the problem?

commit 1d27f6a0 (2022-01-03): fix: Fix crash after Activity is destroyed

Fixes #664