margelo/react-native-vision-camera #665: Not able to use this library, getting so many build issue,  🐛

thisispranavvasudev, 2021-12-20:
### Question

Which node version is suitable for this library?
Which react and react native version is suitable for this library?

### VisionCamera Version

2.10.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

whalemare, 2021-12-21:
What kind of issues?

closed 2021-12-21 by mrousavy

thisispranavvasudev, 2021-12-21:
Please answer to my questions

mrousavy, 2021-12-21:
* Node 14
* All other versions can be seen in the `example/` app - that's what it's for, it's an example. A working example, that works. as an example. it shows how visioncamera works.

mrousavy, 2021-12-21:
https://github.com/mrousavy/react-native-vision-camera/blob/b10ca27daabeb18ec3652ffc64fbbe0fecb298f5/example/package.json#L22-L23

thisispranavvasudev, 2021-12-21:
Okay.. I will try with this then I revert you
Thank you