margelo/react-native-vision-camera #671: 🐛 System err on Samsung device when initialising the camera

HoraceDiesel, 2021-12-22:
### What were you trying to do?

I was trying to initialise the camera on some Samsung devices. This only happens on some samsung devices, only in release builds. The camera can be initialised on debug build completely fine.

my project build.gradle

`
buildscript {
    ext {
        buildToolsVersion = "30.0.2"
        minSdkVersion = 21
        compileSdkVersion = 31
        targetSdkVersion = 30
        ndkVersion = "23.0.7599858"
        androidXCore = "1.0.2"
        firebaseIidVersion = "19.0.1"
        mediaCompatVersion = '1.0.1'
        supportV4Version = '1.0.0'
    }
`

app/build.gradle
`implementation "androidx.camera:camera-core:1.1.0-alpha08"`

### Reproduceable Code

```tsx
const devices = useCameraDevices()
  const device = devices.back
  const isFocused = useIsFocused()

  const onInitialized = useCallback(() => {
    setIsCameraInitialized(true)
  }, [])

if (device == undefined || device == null) return <Loader color={'coolGray.200'} />

return (
    <>
      <TapView
        handleTap={onScreenTapped}
      >
        <Camera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={isFocused}
          ref={camera}
          photo={true}
          onInitialized={onInitialized}
          onError={onError}
          enableZoomGesture={true}
          focusable={true}
        />
      </TapView>
      {
        device !== undefined && (
          <>
            <CaptureButton
              style={styles.captureButton}
              camera={camera}
              onMediaCaptured={onMediaCaptured}
              enabled={isCameraInitialized && isActive}
              flash={flash}
            />
          </>
        )
      }
    </>
  )
}
```

### What happened instead?

No crash, but the app hangs on the loader state, im guessing the device is not available

### Relevant log output

```shell
2021-12-22 11:33:41.134 15881-16923/? W/System.err: java.lang.AbstractMethodError: abstract method "void androidx.camera.extensions.impl.InitializerImpl$OnExtensionsInitializedCallback.onSuccess()"
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at androidx.camera.extensions.impl.-$$Lambda$onPjumpwyrgiPq46UP2QgWyUKsw.run(Unknown Source:2)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at androidx.camera.core.impl.t2.l.b.execute(Unknown Source:0)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at androidx.camera.extensions.impl.InitializerImpl.init(InitializerImpl.java:69)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at c.d.a.f.f(Unknown Source:15)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at c.d.a.b.a(Unknown Source:4)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at c.g.a.b.a(Unknown Source:18)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at c.d.a.f.c(Unknown Source:76)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at c.d.a.f.b(Unknown Source:4)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at com.mrousavy.camera.CameraViewModule$c.n(Unknown Source:109)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at h.w.k.a.a.f(Unknown Source:9)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at kotlinx.coroutines.v0.run(Unknown Source:129)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at kotlinx.coroutines.m2.a.W(Unknown Source:1)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at kotlinx.coroutines.m2.a$c.c(Unknown Source:14)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at kotlinx.coroutines.m2.a$c.m(Unknown Source:28)
2021-12-22 11:33:41.134 15881-16923/? W/System.err:     at kotlinx.coroutines.m2.a$c.run(Unknown Source:0)
2021-12-22 11:33:41.778 23727-23727/? W/adbd: timeout expired while flushing socket, closing
2021-12-22 11:33:43.253 1519-1519/? W/KeyguardUpdateMonitor: invalid subId in handleServiceStateChange()
2021-12-22 11:33:43.253 1519-1724/? E/CscParser: update(): xml file exist
2021-12-22 11:33:43.256 1480-1480/? W/PhoneGlobals: Can't get phone with sub id = -1
2021-12-22 11:33:46.766 1122-1180/? E/BatteryStatsService: Unable to load Power Hal or power.stats HAL
2021-12-22 11:33:47.697 1122-1164/? E/Watchdog: !@Sync: 311 heap: 52 / 72 [2021-12-22 10:33:47.697] sdogWay: softdog FD: 726
2021-12-22 11:33:51.440 32741-504/? W/A: Queue length for executor EventBus is now 11. Perhaps some tasks are too long, or the pool is too small.
2021-12-22 11:33:53.152 1122-2518/? W/CAE: registerCallback(ContextAwareService.java:192) - [regi 01] Mutex is locked for TEST_FLAT_MOTION
2021-12-22 11:33:53.152 1122-2518/? W/CAE: registerCallback(ContextAwareService.java:199) - [regi 04]  com.samsung.android.contextaware.ContextAwareManager$CaListenerDelegate@6401b1f
2021-12-22 11:33:53.152 1122-2518/? W/CAE: getListener(ListenerListManager.java:127) - [getListener1]  com.samsung.android.contextaware.ContextAwareManager$CaListenerDelegate@6401b1f
2021-12-22 11:33:53.152 1122-2518/? W/CAE: getListener(ListenerListManager.java:128) - [getListener2]  com.samsung.android.contextaware.manager.ContextAwareService$Listener@80ecebe
2021-12-22 11:33:53.155 681-681/? E/Sensors: inject_scontext_data: New ssp_data_injection_fd(40)
2021-12-22 11:33:53.161 1122-2518/? W/CAE: registerCallback(ContextAwareService.java:213) - [regi 02] Mutex is unlocked for TEST_FLAT_MOTION
2021-12-22 11:33:53.161 1122-2518/? E/SemContext.CaeProvider: setAttribute() : attribute is null!
2021-12-22 11:33:53.162 1122-2518/? W/Binder: Outgoing transactions from this process must be FLAG_ONEWAY
    java.lang.Throwable
        at android.os.BinderProxy.transact(BinderProxy.java:514)
        at com.samsung.android.hardware.context.ISemContextCallback$Stub$Proxy.getListenerInfo(ISemContextCallback.java:162)
        at com.samsung.android.hardware.context.SemContextService$ListenerManager.notifyListeners(SemContextService.java:1332)
        at com.samsung.android.hardware.context.SemContextService.registerCallback(SemContextService.java:259)
        at com.samsung.android.hardware.context.ISemContextService$Stub.onTransact(ISemContextService.java:156)
        at android.os.Binder.execTransactInternal(Binder.java:1190)
        at android.os.Binder.execTransact(Binder.java:1159)
2021-12-22 11:33:53.421 1122-2518/? W/CAE: unregisterCallback(ContextAwareService.java:234) - [unregi 01] Mutex is locked for TEST_FLAT_MOTION
2021-12-22 11:33:53.421 1122-2518/? W/CAE: getListener(ListenerListManager.java:127) - [getListener1]  com.samsung.android.contextaware.ContextAwareManager$CaListenerDelegate@6401b1f
2021-12-22 11:33:53.421 1122-2518/? W/CAE: getListener(ListenerListManager.java:128) - [getListener2]  com.samsung.android.contextaware.manager.ContextAwareService$Listener@80ecebe
2021-12-22 11:33:53.424 681-681/? E/Sensors: inject_scontext_data: New ssp_data_injection_fd(40)
2021-12-22 11:33:53.430 1122-2518/? W/CAE: unregisterCallback(ContextAwareService.java:255) - [unregi 02] Mutex is unlocked for TEST_FLAT_MOTION
2021-12-22 11:34:17.703 1122-1164/? E/Watchdog: !@Sync: 312 heap: 56 / 72 [2021-12-22 10:34:17.703] sdogWay: softdog FD: 723
2021-12-22 11:34:29.834 3332-3396/? W/System: A resource failed to call close. 
2021-12-22 11:34:29.834 3332-3396/? W/System: A resource failed to call close. 
2021-12-22 11:34:31.481 1122-1346/? W/DeviceStorageMonitorService: updateBroadcasts(/data) oldLevel:0, newLevel:0, seq:1
2021-12-22 11:34:31.482 1122-1346/? W/DeviceStorageMonitorService: updateBroadcasts_filenode(/data) fn_oldLevel:0, fn_newLevel:0, seq:1
```

### Device

Samsung Galaxy S20 5G

### VisionCamera Version

2.9.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

besouro7, 2022-06-17:
Any update on this?

faustaleonardo, 2022-08-02:
Any update on this? happened on my side as well. Infinix, Samsung, realme and Advan.

Akash-Hubspire, 2023-06-09:
Still no update?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy