margelo/react-native-vision-camera #673: ✨ Clear cache

Hirbod, 2021-12-23:
### What feature or enhancement are you suggesting?

Currently, videos and images stored are bloating up my app space. After a couple of hours testing, I already reached 6GB of data. It would be nice to have a clear and clearAll function. ClearAll should just delete everything that the camera had stored and "clear" shall take an path as param to delete the video. Personally speaking, I would love to have a clearAll method to delete everything once the user left a specific screen.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

I think it should be possible to delete those temp files with another filesystem library, but those won't keep track of what belongs to RNVC. I think something internally would be nice.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

Hirbod, 2021-12-23:
Alright, looks like this lib already deletes all files after an app restart. That's actually enough.
I checked with:

```
          const data = await FileSystem.readDirectoryAsync(
            FileSystem.cacheDirectory + '../../tmp/ReactNative'
          );
```

I will need to find out, why the app has 5-6GB of data already, guess other libs already caching a bunch of stuff.

closed 2021-12-23 by Hirbod

digitalbore, 2022-01-28:
Hi @Hirbod  My cache is not cleared after the app restart.

I took 10 photos with **takePhoto** and the app cache increased to 100MB, after closing and reopening the app, the cache is still here.

Is cache clearing enabled by default?

digitalbore, 2022-01-28:
I found this alternative solution:

```
import RNFS from 'react-native-fs';

takePhoto = async () => {
  const photo = await camera.current.takePhoto();
  // Do what you need with the image and then…
  RNFS.unlink(photo.uri); // Remove image from cache
}
```