margelo/react-native-vision-camera #674: ✨ make VISION_CAMERA_DISABLE_FRAME_PROCESSORS=1 configurable via expo-config-plugin

Hirbod, 2021-12-23:
### What feature or enhancement are you suggesting?

In order to be able to disable VISION_CAMERA_DISABLE_FRAME_PROCESSORS=1 for expo, we would need to have config-plugin param for it. Would be nice if you could provide this via the config-plugin.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

None

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2022-01-02:
I don't know how I can write to `GCC_PREPROCESSOR_DEFINITIONS` in the user's `project.pbxproj` file using Expo Config Plugins...

mrousavy, 2022-01-02:
Or to `gradle.properties` for Android

mrousavy, 2022-01-02:
summoning the legend @EvanBacon 🤲

Hirbod, 2022-01-02:
I think this will need some sort of brutforce regexp methods, but I am not sure if the file is "readable"? Evan is most likely the best fit for this question!

Hirbod, 2022-01-04:
Done. #708

PR #708 (2022-01-11): feat: disable frame processors via expo-config-plugin for iOS and Android
## What

This PR adds the ability to disable frame processors for Android and iOS via the shipped expo-config-plugin and also fix shared library compile issues with a dangerous mod. (only activate when you encounter build issues)

Edit: updated to support iOS. This was really painful, because there is basically 0 documentation and I spent like 5-6 hours for such a small task at the end :D

## Usage

Add to your app.config.js/app.json/app.config.json (expo.plugins).

```
    [
      'react-native-vision-camera',
      {
  [... cut]
closed 2022-01-11 by mrousavy