margelo/react-native-vision-camera #675: 🐛 use useFrameProcessor

ronhello, 2021-12-24:
### What were you trying to do?

Enable "frameProcessor", app will be crashed after run Reload JS (from debug menu) or CodePush.restartApp().
Seem like the "FrameProcessor" doesn't cleanup if we reload js bundle.

### What happened instead?

No Crash

### Relevant log output

```shell
2021-12-24 18:47:33.428793+0700 exampleapp[8524:2122342] FrameProcessorBindings: Successfully created worklet!
2021-12-24 18:47:33.495720+0700 exampleapp[8524:2122338] FrameProcessorBindings: Converting worklet to Objective-C callback...
2021-12-24 18:47:33.498141+0700 exampleapp[8524:2122338] FrameProcessorBindings: Frame processor set!
2021-12-24 18:47:33.654409+0700 exampleapp[8524:2122340] <Google> Cannot find an ad network adapter with the name(s): com.google.DummyAdapter. Remember to link all required ad network adapters and SDKs, and set -ObjC in the 'Other Linker Flags' setting of your build target.
2021-12-24 18:47:33.686432+0700 exampleapp[8524:2122339] [native] VisionCamera.didSetProps(_:): Started Session!
libc++abi: terminating with uncaught exception of type facebook::jsi::JSError: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)

Error: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)
    at _f (native)
dyld4 config: DYLD_LIBRARY_PATH=/usr/lib/system/introspection DYLD_INSERT_LIBRARIES=/Developer/usr/lib/libBacktraceRecording.dylib:/Developer/usr/lib/libMainThreadChecker.dylib:/Developer/Library/PrivateFrameworks/DTDDISupport.framework/libViewDebuggerSupport.dylib
terminating with uncaught exception of type facebook::jsi::JSError: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)

Error: Exception in HostFunction: Javascript worklet error

Error: Javascript worklet error
    at _f (native)
    at _f (native)
(lldb)  
```
```

### Device

iPhone

### VisionCamera Version

2.10.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ronhello, 2021-12-24:
Seem like the Frame Processor Plugins are not injected into the VisionCamera JS-Runtime after Js Bundle reloaded.

ronhello, 2021-12-26:
close due to dupplicated https://github.com/mrousavy/react-native-vision-camera/issues/626

closed 2021-12-26 by ronhello

mrousavy, 2021-12-26:
Hm, I see. This probably has to be moved to the same setup code as the JSI funcs

ronhello, 2021-12-27:
> Hm, I see. This probably has to be moved to the same setup code as the JSI funcs

I removed the line ```[FrameProcessorPluginRegistry markInvalid];``` from ```initWithBridge``` and it's working well on iOS.
Now I'm facing the same problem with https://github.com/mrousavy/react-native-vision-camera/issues/570 and https://github.com/mrousavy/react-native-vision-camera/issues/500 on iOS.

PR #684 (2021-12-30): fix: Re-install FP plugins on bundle reload

## What

Re-install/initialize all Frame processor plugins on bundle reload

## Changes

## Tested on

## Related issues

* Fixes #675