margelo/react-native-vision-camera #683: ❓ How do I know when focus() has finished?

sandlands, 2021-12-28:
### Question

In the API I see there is a [focus](https://mrousavy.com/react-native-vision-camera/docs/api/classes/Camera#focus) method, which focuses the camera to a certain point. It returns a promise, but it looks like the promise is resolved before the focus procedure has finished. Is that true? Is there any way to know when focusing has finished (with a callback for example)?

It looks like the focus procedure duration is different on different phones (and probably in different scenes as well), so it would be nice to know when the lens has focused and is ready to take a (sharp) picture.

Thanks in advance!

### VisionCamera Version

2.9.4

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-12-30:
Hey! Good question, I think there is no callback to find out when focussing has been successful, but I'll take a look soon!

sandlands, 2022-01-10:
It looks like on Android the promise is only resolved when focussing has finished. On iOS, it's immediately resolved.

nishain-de-silva, 2023-05-14:
Hi I checked the direct source code after some I realized that in IOS focusing occur in continuous auto focus mode so focusing is never end on IOS so it make sense why promise is resolved immediately in IOS because there is no end for focusing after all. I suppose you can capture the image after 2 - 3 seconds.
But in android auto focus happens once and promise is resolved afterwards l.

mrousavy, 2024-04-22:
Hey - so the `focus()` function will now only resolve once focus has happened. It can also reject/throw if focus was canelled. So this is now implemented in V4.

closed 2024-04-22 by mrousavy