margelo/react-native-vision-camera #687: ❓Meaning of Frame's planesCount and bytesPerRow (iOS)

sandlands, 2021-12-31:
### Question

Hi there,

Thanks for maintaining this great library. I'm trying to implement a FrameProcessor for iOS (at first), with a final goal to call some OpenCV functions. In order to be able to convert the `CMSampleBufferRef` to a `cv::Mat`. I need to know the 'line stride' of the buffer (see definition [here](https://medium.com/@oleg.shipitko/what-does-stride-mean-in-image-processing-bba158a72bcd)).

I see the following fields:
```
Frame: {"isValid":true,"width":1504,"height":1128,"bytesPerRow":2,"planesCount":2}
```

1) I would expect `bytesPerRow` to be the "line stride" (pixel_depth * width + line padding) and the `planesCount` to be the "pixel_depth". Am I wrong?

2) Do you maybe know the **PixelFormat** (RGB(A), BGR(A), ...) as well?

Thanks in advance!
Sander

- I'm using version 2.9.4 (no yet upgraded to 2.10.0 because of [this issue](url))
- iPhone XS

### VisionCamera Version

2.9.4

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-12-31:
Hi!

Yeah, that seems to be a fair assumption. You can also get those props using the CMSampleBuffer's methods, that's how I do it under the hood anyways.

Actually in the latest VisionCamera version you can inspect a format's `pixelFormat` field and select the format that has the desired pixel format. (https://github.com/mrousavy/react-native-vision-camera/commit/c3039c94c69805edff7070a573c1d9d5612f7d6a)

sandlands, 2021-12-31:
Thanks for your reply. It seems that there is a bug in [this line](https://github.com/mrousavy/react-native-vision-camera/blob/b7bfa5ef0ad9a1c0add3d3508d7a4e0c65d2f6da/ios/Frame%20Processor/FrameHostObject.mm#L72).

`CVPixelBufferGetPlaneCount ` should be `CVPixelBufferGetBytesPerRow`

mrousavy, 2021-12-31:
Ah sh*t you're right!

mrousavy, 2022-01-02:
Fixed! Check out the latest update

mrousavy, 2022-01-03:
Closing since the issue has been resolved, let me know if your question is not yet answered. Also, if you have permission, I'd love to see the Frame Processor Plugin you've built once you're done 😄

closed 2022-01-03 by mrousavy