margelo/react-native-vision-camera #690: How to set focus once to be the closest possible (ex for macro photography)❓

cyprian, 2021-12-31:
### Question

Is there a way to set focus on a "device" to be fixed (unchanged) and be the closest possible?

How would I approach that? 

P.S. Thank you for offering this great library! 

### What I tried

I know that the Camera API offers "[Focus](https://mrousavy.com/react-native-vision-camera/docs/guides/focusing)" function but that is based on a given Point (x.y).

### VisionCamera Version

2.10.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-01-03:
Hey!

Not sure about that, focus seems to turn off after a while again. Perhaps a workaround would be to repeatedly call it?

mrousavy, 2023-09-30:
Hey - this is not a question but more like a feature request. Currently you can only focus on x,y screen coords, but I could add a feature to focus on a specific ƒ length. Create a feature request so I can track this for V3 :)

closed 2023-09-30 by mrousavy

Lovishchugh01, 2025-10-06:
any update on this?