margelo/react-native-vision-camera #702: A problem occurred evaluating project ':react-native-vision-camera'🐛

Bamorem, 2022-01-03:
### What were you trying to do?

I was running my app without issue with the version 2.10.0, but since the update of 2.11.0 the build failed with an exception.

on the /react-native-vision-camera/android/build.gradle' line: 8
it's looking for the file /react-native-vision-camera/node_modules/react-native/ReactAndroid/gradle.properties 
which doesn't exist (No such file or directory).

I tried to clean the caches but still happening
[image]

### Reproduceable Code

```tsx
"react-native-vision-camera": "^2.11.0",
```

### What happened instead?

FAILURE: Build failed with an exception.

### Relevant log output

```shell
FAILURE: Build failed with an exception.

* Where:
Build file '/Users/yohanbalassiano/Sites/Deelt/app/deelt/node_modules/react-native-vision-camera/android/build.gradle' line: 8

* What went wrong:
A problem occurred evaluating project ':react-native-vision-camera'.
> /Users/yohanbalassiano/Sites/Deelt/app/deelt/node_modules/react-native-vision-camera/node_modules/react-native/ReactAndroid/gradle.properties (No such file or directory)
```

### Device

OnePlus Nord

### VisionCamera Version

2.11.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-01-03:
Hi, thank you for the report.

That bug has already been reported in a comment before, it's because the path to the node_modules folder seems to be wrong. I will take a look soon, for now you can hotpatch it using patch-package :)

mrousavy, 2022-01-03:
Okay had some free time to investigate this - can you please try if #704 works for you?

closed 2022-01-03 by mrousavy

jensdev, 2022-01-03:
Hi @mrousavy, I had the same issue. When using the `fix/monorepo-support` I get the following.

```
FAILURE: Build failed with an exception.

* Where:
Build file '/home/jens/projects/craftzing/extraHorizon/fibricheck_react_native/node_modules/react-native-vision-camera/android/build.gradle' line: 24

* What went wrong:
A problem occurred evaluating project ':react-native-vision-camera'.
> /home/jens/projects/craftzing/extraHorizon/fibricheck_react_native/node_modules/react-native-vision-camera/node_modules/react-native/ReactAndroid/gradle.properties (No such file or directory)
```

If I change line 19
```diff
-def nodeModules = findNodeModules(projectDir)
+def nodeModules = findNodeModules(rootDir)
```

It does build

mrousavy, 2022-01-03:
Maybe your `node_modules/react-native-vision-camera` folder also has another `node_modules/` folder inside? if yes, delete that. 

I'll try your suggested solution if that works here ...

mrousavy, 2022-01-03:
Let's see if #706 builds in the CI.

ussef11, 2024-02-01:
Thanks  @jensdev  its work for me

PR #704 (2022-01-03): feat: Monorepo support

## What

Finds `node_modules/` path dynamically.

## Changes

## Tested on

## Related issues

* Fixes #702