margelo/react-native-vision-camera #707: 🐛 VisionCamera_disableFrameProcessors not working?

Hirbod, 2022-01-03:
### What were you trying to do?

I am not sure if this is my mistake or if there is another small bug, but I tried to patch-package your libs gradle.properties and I also changed gradle.properties inside /android, adding

`VisionCamera_disableFrameProcessors=true` but the console is still compiling frame processors and I have this log in my console:

VisionCamera: Frame Processors are enabled! Building C++ part...

### What happened instead?

VisionCamera: Frame Processors are enabled! Building C++ part...

### Relevant log output

```shell
VisionCamera: Frame Processors are enabled! Building C++ part...
```

### Device

Redmi Note

### VisionCamera Version

2.11.0

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-01-03:
You need to add

```
disableFrameProcessors=true
```

, without the `VisionCamera_` prefix

Hirbod, 2022-01-03:
VisionCamera: Frame Processors are disabled because the user explicitly disabled it ('disableFrameProcessors=true'). C++ part will not be built.

🤦 

Thanks!

Hirbod, 2022-01-03:
Huge compile difference. Now I need to find a way for iOS, but not bothering you anymore for now :D

closed 2022-01-03 by Hirbod

mrousavy, 2022-01-03:
Would appreciate it if you could contribute and add a `disableFrameProcessors` parameter to the `withVisionCamera` config plugin so it's also easy for Expo users 🙏 

I gotta get back to work now 😅

Hirbod, 2022-01-03:
@mrousavy that is what I am on it right now :)