margelo/react-native-vision-camera #709: 🐛 takePhoto doesn't work with Android SDK 26

savv, 2022-01-03:
### What were you trying to do?

I literally wanted to take a picture :-)

### What happened instead?

An error is thrown: [unknown/unknown: [unknown/unknown] Not in application's main thread]

### Device

Android Emulator for Api 26

### VisionCamera Version

2.11.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-01-03:
Hmm, is there a stack trace? I'm assuming it's this call, but `coroutineScope` should be the main application Dispatcher..?

https://github.com/mrousavy/react-native-vision-camera/blob/f4d58412a659fc926748d16351430c088c998a3d/android/src/main/java/com/mrousavy/camera/CameraView%2BTakePhoto.kt#L72

savv, 2022-01-03:
Stack trace below:
```
01-03 16:32:39.395  1432  1617 W audio_hw_generic: Not supplying enough data to HAL, expected position 5011126 , only wrote 4880880
01-03 16:32:39.508 12022 12033 I zygote  : Background concurrent copying GC freed 113911(6MB) AllocSpace objects, 0(0B) LOS objects, 7% free, 69MB/75MB, paused 1.183ms total 124.999ms
01-03 16:32:41.104 12022 12022 W unknown:ReactNative: Can't cancel already finished gesture. Is a child View trying to start a gesture from an UP/CANCEL event?
01-03 16:32:41.141  1433  8790 V EmulatedCamera_Camera: getCameraInfo
01-03 16:32:41.141  1433  8790 V EmulatedCamera_BaseCamera: getCameraInfo
01-03 16:32:41.145 12022 12651 I CameraView: takePhoto() called, but falling back to Snapshot because 1 use-case is already occupied.
01-03 16:32:41.147 12022 12651 W System.err: java.lang.IllegalStateException: Not in application's main thread
01-03 16:32:41.153 12022 12651 W System.err: 	at androidx.core.util.Preconditions.checkState(Preconditions.java:151)
01-03 16:32:41.154 12022 12651 W System.err: 	at androidx.camera.core.impl.utils.Threads.checkMainThread(Threads.java:50)
01-03 16:32:41.154 12022 12651 W System.err: 	at androidx.camera.view.PreviewView.getBitmap(PreviewView.java:482)
01-03 16:32:41.154 12022 12651 W System.err: 	at com.mrousavy.camera.CameraView_TakeSnapshotKt$takeSnapshot$2.invokeSuspend(CameraView+TakeSnapshot.kt:25)
01-03 16:32:41.154 12022 12651 W System.err: 	at com.mrousavy.camera.CameraView_TakeSnapshotKt$takeSnapshot$2.invoke(Unknown Source:8)
01-03 16:32:41.154 12022 12651 W System.err: 	at com.mrousavy.camera.CameraView_TakeSnapshotKt$takeSnapshot$2.invoke(Unknown Source:4)
01-03 16:32:41.154 12022 12651 W System.err: 	at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:89)
01-03 16:32:41.154 12022 12651 W System.err: 	at kotlinx.coroutines.CoroutineScopeKt.coroutineScope(CoroutineScope.kt:264)
01-03 16:32:41.154 12022 12651 W System.err: 	at com.mrousavy.camera.CameraView_TakeSnapshotKt.takeSnapshot(CameraView+TakeSnapshot.kt:16)
01-03 16:32:41.154 12022 12651 W System.err: 	at com.mrousavy.camera.CameraView_TakePhotoKt$takePhoto$2.invokeSuspend(CameraView+TakePhoto.kt:22)
01-03 16:32:41.154 12022 12651 W System.err: 	at com.mrousavy.camera.CameraView_TakePhotoKt$takePhoto$2.invoke(Unknown Source:8)
01-03 16:32:41.155 12022 12651 W System.err: 	at com.mrousavy.camera.CameraView_TakePhotoKt$takePhoto$2.invoke(Unknown Source:4)
01-03 16:32:41.155 12022 12651 W System.err: 	at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:89)
01-03 16:32:41.155 12022 12651 W System.err: 	at kotlinx.coroutines.CoroutineScopeKt.coroutineScope(CoroutineScope.kt:264)
01-03 16:32:41.155 12022 12651 W System.err: 	at com.mrousavy.camera.CameraView_TakePhotoKt.takePhoto(CameraView+TakePhoto.kt:19)
01-03 16:32:41.155 12022 12651 W System.err: 	at com.mrousavy.camera.CameraViewModule$takePhoto$1.invokeSuspend(CameraViewModule.kt:85)
01-03 16:32:41.155 12022 12651 W System.err: 	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
01-03 16:32:41.155 12022 12651 W System.err: 	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
01-03 16:32:41.155 12022 12651 W System.err: 	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
01-03 16:32:41.155 12022 12651 W System.err: 	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
01-03 16:32:41.156 12022 12651 W System.err: 	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
01-03 16:32:41.156 12022 12651 W System.err: 	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
01-03 16:32:41.176 12022 13313 E ReactNativeJS: 'Failed to take photo!', { [unknown/unknown: [unknown/unknown] Not in application's main thread]
01-03 16:32:41.176 12022 13313 E ReactNativeJS:   name: 'unknown/unknown',
01-03 16:32:41.176 12022 13313 E ReactNativeJS:   _code: 'unknown/unknown',
01-03 16:32:41.176 12022 13313 E ReactNativeJS:   _message: '[unknown/unknown] Not in application\'s main thread',
01-03 16:32:41.176 12022 13313 E ReactNativeJS:   _cause: undefined }
01-03 16:32:43.349  1724  1748 E memtrack: Couldn't load memtrack module
01-03 16:32:43.349  1724  1748 W android.os.Debug: failed to get memory consumption info: -1
```

savv, 2022-01-03:
So it looks like it's not that call. It's switching to using snapshots. Maybe it's an issue specific to the emulator's camera?

mrousavy, 2022-01-03:
Oh I've seen that before I think - takeSnapshot is apparently not being called on the Main Thread. hmm

savv, 2022-01-03:
I believe it's this one: https://github.com/mrousavy/react-native-vision-camera/issues/547

Sorry for double filing. I thought this merited a separate issue as it seems more serious. I guess it depends on whether `fallbackToSnapshot` is common or not - as it would make snapshots a key feature (only way to take photos) for those phones that need it.

mrousavy, 2022-01-03:
I can't believe Android actually made phones that didn't support video and photo capture in the same Camera View.

savv, 2022-01-03:
I tested on a samsung galaxy s7 edge with Android 8.0 (api 26) and I was able to take photos, so this is specific to snapshots, not the API version.

savv, 2022-01-17:
Closing this as I was able to resolve it by disabling video, in which case fallbackToSnapshot is not needed.

closed 2022-01-17 by savv

mateusschenato, 2022-03-29:
Any solution ?

deflexable, 2022-05-16:
@bekatd just remove the video props when taking photo or remove the photo props when recording a video