margelo/react-native-vision-camera #713: Is there a way to turn off capture sound❓

PABourdais, 2022-01-04:
### Question

Is there a way to turn off capture sound when pictures is taken ?

### What I tried

I know that it was possible on react-native-camera

### VisionCamera Version

2.11.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-01-04:
Hi! Does it make a sound on Android?

mrousavy, 2022-01-04:
Also, duplicate of #713. Let's continue the discussion over there

closed 2022-01-04 by mrousavy

mrousavy, 2023-08-21:
Hey! I just implemented this in V3! :) https://github.com/mrousavy/react-native-vision-camera/pull/1702