margelo/react-native-vision-camera #714: Is there any way to change to AVCaptureVideoDataOutput pixelFormat❓

nicoache1, 2022-01-04:
### Question

Hi im trying to create a custom Frame processor in order to run TensorFlow Lite [MoveNet](https://blog.tensorflow.org/2021/08/pose-estimation-and-classification-on-edge-devices-with-MoveNet-and-TensorFlow-Lite.html) 
One of the issues im experiencing is that the buffer of the frame it comes with a format `kCVPixelFormatType_420YpCbCr8BiPlanarVideoRange` but Tensorflow needs the format `kCVPixelFormatType_32ARGB`. I solve that adding settings to the videoOutput in the `CameraView+AVCaptureSession` specifing the pixel format that I need.

**The question is more like if there are any plans to gave the user support to change the format to specific one?**
I think in some cases like this its pretty valuable because it's better to specify this to the AVCaptureVideoDataOutput for example than make the whole conversion.

Thanks!

### VisionCamera Version

2.11.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-01-04:
Hi! 

You can iterate over `device.formats` to find a format with the desired `.pixelFormat`, and then select that by passing it to the `<Camera>` component :)
See [pixelFormat](https://mrousavy.com/react-native-vision-camera/docs/api/interfaces/CameraDeviceFormat#pixelformat)

nicoache1, 2022-01-04:
Thanks for your quick response @mrousavy 

> Hi!
> 
> You can iterate over `device.formats` to find a format with the desired `.pixelFormat`, and then select that by passing it to the `<Camera>` component :) See [pixelFormat](https://mrousavy.com/react-native-vision-camera/docs/api/interfaces/CameraDeviceFormat#pixelformat)

I tried that to find if any format is available to me but what I get its `420f`, `420v` or `x420` that all of them are `YpCbCr` formats and I need an another kind of representation as I mention. I dont know if that is because the device itself of its related to this [file](https://github.com/mrousavy/react-native-vision-camera/blob/1249cde3371ad6095528aee947b9d779135e640a/src/PixelFormat.ts).

I know that exists some conversions using `Accelerate` and doing some math but it's more easier if you let that work to `AVCaptureVideoDataOutput` just putting in the settings the desired format.
My solution was just add these lines in the class `CameraView+AVCaptureSession`line: 135 and let they do the job.

```
let outputSettings: [String: Any] = [String(describing: kCVPixelBufferPixelFormatTypeKey): NSNumber(value: kCVPixelFormatType_32BGRA)]
videoOutput!.videoSettings = outputSettings
```

What do you think about this scenario?

mrousavy, 2022-01-04:
That's very interesting. If it is easily configureable then VisionCamera should provide that as a prop. I guess that depends on the Format though, maybe `pixelFormat` is not reporting all formats correctly?

nicoache1, 2022-01-04:
Yes that was one of the things that I think about.

I dig a deep more in iOS code trying to understand and I think about `pixelFormat` is ok, it gets the device pixelFormat using `CMFormatDescriptionGetMediaSubType`.
The thing that maybe its more accurate to say its that this pixelFormat its related to the videoInput (You get that in line:72 in `CameraView+AVCaptureSession`, but what im trying to being able to configure its the videoOutput in line:138, maybe its not strictly related one an another. 

I guess  that if you dont specify one in the videoOutput it gives you the `Frame` in the same pixelFormat as your videoInput, but if you put something in the `videoOutput.videoSettings` then it transforms that.

mrousavy, 2022-01-06:
Ah I think I know why. I was having a 1:1 session with an Apple developer on last WWDC, and I remember talking about exactly this problem. The output pixel format can be customized, but if you choose anything different than the format's pixel format, it will be an internal conversion. So it is not natively streaming in ARGB_8888, but rather streaming in e.g. 420v, and then doing a conversion to ARGB_8888. There is no native ARGB_8888 format.

nicoache1, 2022-01-06:
Interesting. That makes more sense with the fact that when you obtain the formats of the cameras you always receive 420X.
I dont know what do you think but maybe its wort it to add an optional iOS prop ( at least for the moment, I dont know how Android tackle this yet ) that gave the possibility to access this internal conversion.