margelo/react-native-vision-camera #716: 🐛  Build Failed in new app created with react-native init

firofame, 2022-01-05:
### What were you trying to do?

run the app on ios simulator/device after installing react-native-vision-camera

### What happened instead?

Build failed with "Cannot convert value of type '[AnyHashable : Any]' to expected argument type '[String : Any]'"

### Relevant log output

```shell
npx react-native info
info Fetching system and libraries information...
System:
    OS: macOS 11.6.2
    CPU: (8) arm64 Apple M1
    Memory: 142.83 MB / 8.00 GB
    Shell: 5.8 - /bin/zsh
  Binaries:
    Node: 17.0.1 - /opt/homebrew/bin/node
    Yarn: 1.22.17 - /opt/homebrew/bin/yarn
    npm: 8.1.0 - /opt/homebrew/bin/npm
    Watchman: 2021.11.15.00 - /opt/homebrew/bin/watchman
  Managers:
    CocoaPods: 1.11.2 - /opt/homebrew/bin/pod
  SDKs:
    iOS SDK:
      Platforms: iOS 14.4, DriverKit 20.2, macOS 11.1, tvOS 14.3, watchOS 7.2
    Android SDK:
      API Levels: 29, 31
      Build Tools: 29.0.2, 31.0.0
      System Images: android-29 | Intel x86 Atom_64, android-31 | Google APIs ARM 64 v8a
      Android NDK: Not Found
  IDEs:
    Android Studio: 2020.3 AI-203.7717.56.2031.7784292
    Xcode: 12.4/12D4e - /usr/bin/xcodebuild
  Languages:
    Java: 1.8.0_292 - /usr/bin/javac
  npmPackages:
    @react-native-community/cli: Not Found
    react: 17.0.2 => 17.0.2 
    react-native: 0.66.4 => 0.66.4 
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found
```

### Device

iPhone 12 (iOS 14.4)

### VisionCamera Version

2.11.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

firofame, 2022-01-05:
XCode 12.4, Swift version 5.3.2

firofame, 2022-01-06:
/Users/firozahmed/Desktop/MyApp/node_modules/react-native-vision-camera/ios/CameraView+RecordVideo.swift:135:70: Cannot convert value of type '[AnyHashable : Any]' to expected argument type '[String : Any]'

@mrousavy  this is the exact error I get. I wonder if it's just me

champikamendis, 2022-01-06:
I got the same error

firofame, 2022-01-06:
@champikamendis what version of Xcode are you using?

champikamendis, 2022-01-06:
XCode 12.4

mrousavy, 2022-01-06:
Okay looks like they changed the type of that value. If I cast it, it breaks for other users 😅

firofame, 2022-01-06:
there is one more build error. just an fyi

/Users/firozahmed/Desktop/MyApp/node_modules/react-native-vision-camera/ios/CameraView+RecordVideo.swift:267:26: Type of expression is ambiguous without more context

champikamendis, 2022-01-06:
Can @mrousavy tell me what to change in my code? I have two errors when building like @firofame faced.

1. "Cannot convert value of type '[AnyHashable : Any]' to expected argument type '[String : Any]'"
2. "Type of expression is ambiguous without more context"
Can you tell what are the changes to be made?

HERMESMATEUS, 2022-01-06:
I got the same issue, my xcode version is 12.4 with "react": "16.13.1", "react-native": "0.63.3"

node_modules/react-native-vision-camera/ios/CameraView+RecordVideo.swift:135:70: Cannot convert value of type '[AnyHashable : Any]' to expected argument type '[String : Any]'

node_modules/react-native-vision-camera/ios/CameraView+RecordVideo.swift:267:26: Type of expression is ambiguous without more context

firofame, 2022-01-07:
vision-camera works fine with XCode 13. unfortunately, I am stuck with XCode 12.4 since I have some dependency on Swift 5.3.

@mrousavy is there an older version that works with XCode 12.4?

brianangulo, 2022-01-11:
Having the same issue on xcode 12.5

ReyGuez, 2022-01-13:
Having the same issue on Xcode 13, any solution?

Borhanof, 2022-01-13:
Having the same issue please help me out how to fix it

Navipro70, 2022-01-14:
Same issue, for XCode 12.5 (also test with XCode 13.2, also unsuccessful because of new errors), and RN 66.4

mrousavy, 2022-01-19:
On my end it builds, running Xcode 13.2.1 (13C100)

Navipro70, 2022-01-19:
@mrousavy 
Sorry, errors described at my previous message are not related to Xcode version (see screenshot with errors). I'm typecast places with errors in _VisionCamera_ swift module on Xcode 12.5 and it starts building, but I got another error's, same as on Xcode 13.2:
[image]

hungdev, 2022-01-21:
i get same issue

Aris-ngoy, 2022-01-26:
> Okay looks like they changed the type of that value. If I cast it, it breaks for other users 😅

what do you propose?

mrousavy, 2022-01-27:
I propose that I fix the issue once I also get this issue. For me, on Xcode 13, everything builds fine so far. I also mentioned all of the versions (e.g. Swift) in the Troubleshooting section.

GKyomen, 2022-01-27:
same error here. Any prevision on fixing?

davidshariff, 2022-02-03:
Same issue: Xcode 12.5, Swift 5.3

hungdev, 2022-02-07:
I dont understand what's probblem. But i do some steps and it can run successfully.
Delete node module and remove yarn.lock ( use yarn instead of npm, i used npm before ). And change "react-native-vision-camera": "2.9.4", ( delete ^ in front 2.9.4). and delete Podfile.lock then `npx pod-install`
Rerun and it worked for me.
Xcode Version 12.5
"react-native": "0.64.2"

john-christian-melendres, 2022-03-14:
I also encounter the same issue, I'm using Xcode 12.4 and Swift 5.3.2

john-christian-melendres, 2022-03-16:
I'm did not encounter the error anymore after updating to Xcode 13.3

mrousavy, 2022-03-16:
Great!

closed 2022-03-16 by mrousavy

firofame, 2022-03-16:
@mrousavy 
"Troubleshooting" page says "VisionCamera requires a minimum Swift version of 5.2", This needs to be changed to XCode 13. 

many repositories have a minimum requirement section in the readme so that people with older version do not waste their time.

firofame, 2022-03-16:
Sir, I respect & appreciate your work very much.

swift 5.2 was shipped with XCode 11.4. https://swiftversion.net/

Everyone was using XCode 12+. so, they probably had swift 5.2+ and still had the same error.

I trust you 100% when you say you were using XCode 12 when you originally created this project, but is there a chance the minimum version could have changed for the latest version of VisionCamera?

mrousavy, 2022-03-16:
Thanks! Yeah, I guess we could add an additional point to the TROUBLESHOOTING guide that states "Make sure you're using Xcode 13 or higher". just to be sure

marinne, 2022-03-17:
If you encounter this issue and cannot upgrade to Xcode 13, you can try version 2.10.0, which, during my tests, is the latest version currently that doesn't have this issue in Xcode 12. It's not quite a solution but 2.10.0 may already suit your needs.

hushaudio, 2022-07-25:
> If you encounter this issue and cannot upgrade to Xcode 13, you can try version 2.10.0, which, during my tests, is the latest version currently that doesn't have this issue in Xcode 12. It's not quite a solution but 2.10.0 may already suit your needs.

thank you this worked instantly