margelo/react-native-vision-camera #721: 🐛 useCameraDevices doesn‘t work

jianjiax, 2022-01-07:
### What were you trying to do?

run the app on android device after installing react-native-vision-camera

### Reproduceable Code

```tsx
const devices = useCameraDevices('wide-angle-camera')
```

### What happened instead?

doesn't work

### Relevant log output

```shell
ERROR  Error: Invalid hook call. Hooks can only be called inside of the body of a function component. This could happen for one of the following reasons:
1. You might have mismatching versions of React and the renderer (such as React DOM)
2. You might be breaking the Rules of Hooks
3. You might have more than one copy of React in the same app
See https://reactjs.org/link/invalid-hook-call for tips about how to debug and fix this problem.
```

### Device

android 10

### VisionCamera Version

2.11.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-01-07:
That is still not reproduceable!
The code obviously works for me in the example app.

closed 2022-01-07 by mrousavy

chandumatrix004, 2022-01-16:
Can we use the react-native-vision-camera along with expo-camera @mrousavy ?? I have already developed expo-camera and I want to add Vision camera's features to it. Can I do that??

saqibkafeel, 2023-08-13:
> That is still not reproduceable! The code obviously works for me in the example app.

this error occur in class component instead of functional component
the issue will reproduce