margelo/react-native-vision-camera #725: possibility to configure compression for takePhoto

viljark, 2022-01-10:
I would like to have a compressed image as quickly as possible without having to invoke my own image compression methods on the original image, which takes quite long time on some slower android devices. At the moment I am using [expo-image-manipulator](https://docs.expo.dev/versions/latest/sdk/imagemanipulator/#imagemanipulatormanipulateasyncuri-actions-saveoptions) to compress the original file.

If this is possible on the native APIs, then it would be great if we could have this option available in react-native-vision-camera aswell.

_Originally posted by @viljark in https://github.com/mrousavy/react-native-vision-camera/discussions/722_

FDiskas, 2022-11-17:
related to PR https://github.com/mrousavy/react-native-vision-camera/pull/1201

mrousavy, 2023-09-30:
Unfortunately there's no way to do this on iOS, so I can't really implement it yet...

mrousavy, 2023-09-30:
actually let's leave this open

omerts, 2024-03-09:
We are using an react-native-image-resizer, to change the size and quality of the image, maybe it can be integrated into this lib