margelo/react-native-vision-camera #727: 🐛 The given Camera device could not be found for use-case binding!

Angelk90, 2022-01-10:
### What were you trying to do?

I clone the project: https://github.com/mrousavy/react-native-vision-camera

I tried to run the example file, after many problems, I was able to run it on an android physical device.

### What happened instead?

[image]

[image]

Main camera appears to be out of order, screen all black.

The secondary camera (the one for video calls for example) seems to be working.

When I try to take a photo with the secondary camera, I get this:

[image]

You don't even see the HDR icon.

[image]

Everything seems to go very slow, opening the app.

@mrousavy : What am I doing wrong?

### Device

motorola one (xt1941-4)

### VisionCamera Version

2.11.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Angelk90, 2022-01-10:
@mrousavy :

Tablet Samsung Tab A(2016):

It does not work of the two cameras.

[image]

Motorola one fusion+:

Main camera does not work.
[image]

mrousavy, 2022-01-10:
This is indeed very weird, possibly a CameraX bug. Can you share the adb logs?

Angelk90, 2022-01-10:
@mrousavy :  It doesn't work with any of the physical android devices I've tried.

```
01-10 16:19:05.292  3628  3686 I ReactNativeJS: Re-rendering camera page with active camera. Device: "back (0)" (3840x2160 @ 60fps)
01-10 16:19:05.340  3628  3628 I CameraView: Configuring session...
01-10 16:19:05.341  3628  3628 I CameraView: Configuring session with Camera ID 0 and custom format...
01-10 16:19:05.350  3628  3628 I CameraView: Using custom format - photo: 3840x2160, video: 3840x2160 @ 60 FPS
01-10 16:19:05.351  3628  3628 I CameraView: Setting AE_TARGET_FPS_RANGE to 60-60, and SENSOR_FRAME_DURATION to 0
01-10 16:19:05.352  3628  3628 D Recorder: Video source has transitioned to state: INACTIVE
01-10 16:19:05.355  3628  3628 D DeferrableSurface: surface closed,  useCount=1 closed=true androidx.camera.core.SurfaceRequest$2@51e30e1
01-10 16:19:05.355  3628  3628 D DeferrableSurface: surface closed,  useCount=1 closed=true androidx.camera.core.SurfaceRequest$2@c1c8492
01-10 16:19:05.355  3628  3628 D DeferrableSurface: surface closed,  useCount=1 closed=true androidx.camera.core.impl.ImmediateSurface@b913dbf
01-10 16:19:05.356  3628  3756 D Camera2CameraImpl: {Camera@2acd80c[id=1]} Use cases [androidx.camera.core.Preview-b353fc59-4218-4dfb-8e33-97d4ddb52195142067502, androidx.camera.video.VideoCapture-ae066d4d-c72d-4c51-8e7c-6a4be45ec6d688343247, androidx.camera.core.ImageAnalysis-3cd47eb4-249d-483e-9c11-772505a2ce2232142428] now DETACHED for camera
01-10 16:19:05.356  3628  3628 I CameraView: Adding VideoCapture use-case...
01-10 16:19:05.356  3628  3756 D UseCaseAttachState: All use case: [] for camera: 1
01-10 16:19:05.357  3628  3756 D Camera2CameraImpl: {Camera@2acd80c[id=1]} Resetting Capture Session
01-10 16:19:05.361  3628  3756 D Camera2CameraImpl: {Camera@2acd80c[id=1]} Releasing session in state OPENED
01-10 16:19:05.361  3628  3756 D UseCaseAttachState: Active and attached use case: [] for camera: 1
01-10 16:19:05.363  3628  3628 I CameraView: Tried to add photo use-case (`photo={true}`) but the Camera device only supports a single use-case at a time. Falling back to Snapshot capture.
01-10 16:19:05.363  3628  3628 I CameraView: Adding ImageAnalysis use-case...
01-10 16:19:05.364  3628  3756 D UseCaseAttachState: Active and attached use case: [] for camera: 1
01-10 16:19:05.364  3628  3628 I CameraView: Attaching 2 use-cases...
01-10 16:19:05.365  3628  3756 D Camera2CameraImpl: {Camera@2acd80c[id=1]} Closing camera.
01-10 16:19:05.366  3628  3756 D Camera2CameraImpl: {Camera@2acd80c[id=1]} Transitioning camera internal state: OPENED --> CLOSING
01-10 16:19:05.367  3628  3756 D CameraStateRegistry: Recalculating open cameras:
01-10 16:19:05.367  3628  3756 D CameraStateRegistry: Camera                                       State                 
01-10 16:19:05.367  3628  3756 D CameraStateRegistry: -------------------------------------------------------------------
01-10 16:19:05.367  3628  3756 D CameraStateRegistry: Camera@2acd80c[id=1]                         CLOSING               
01-10 16:19:05.367  3628  3756 D CameraStateRegistry: Camera@39bde0[id=0]                          UNKNOWN               
01-10 16:19:05.367  3628  3756 D CameraStateRegistry: -------------------------------------------------------------------
01-10 16:19:05.367  3628  3756 D CameraStateRegistry: Open count: 1 (Max allowed: 1)
01-10 16:19:05.368  3628  3756 D CameraStateMachine: New public camera state CameraState{type=CLOSING, error=null} from CLOSING and null
01-10 16:19:05.368  3628  3756 D CameraStateMachine: Publishing new public camera state CameraState{type=CLOSING, error=null}
01-10 16:19:05.369  3628  3756 D Camera2CameraImpl: {Camera@2acd80c[id=1]} Resetting Capture Session
01-10 16:19:05.371  3628  3756 D Camera2CameraImpl: {Camera@2acd80c[id=1]} Releasing session in state CLOSING
01-10 16:19:05.373  3628  3756 D CaptureSession: onSessionFinished()
01-10 16:19:05.374  3222  3222 I CameraService: Camera : Shutting down 3628 score 0
01-10 16:19:05.375  3222  3222 W libc    : Unable to set property "camera.mot.is.client.foreground" to "1": error code: 0x18
01-10 16:19:05.376  3460 12630 I QCameraHalWatchdog: Starting Watchdog Thread...
01-10 16:19:05.378  3628  3628 D VideoCapabilities: profile = CamcorderProfileProxy{duration=30, quality=8, fileFormat=2, videoCodec=2, videoBitRate=42000000, videoFrameRate=30, videoFrameWidth=3840, videoFrameHeight=2160, audioCodec=3, audioBitRate=320000, audioSampleRate=48000, audioChannels=2}
01-10 16:19:05.378  3628  3628 D VideoCapabilities: profile = CamcorderProfileProxy{duration=30, quality=6, fileFormat=2, videoCodec=2, videoBitRate=17000000, videoFrameRate=30, videoFrameWidth=1920, videoFrameHeight=1080, audioCodec=3, audioBitRate=320000, audioSampleRate=48000, audioChannels=2}
01-10 16:19:05.379  3628  3628 D VideoCapabilities: profile = CamcorderProfileProxy{duration=30, quality=5, fileFormat=2, videoCodec=2, videoBitRate=10000000, videoFrameRate=30, videoFrameWidth=1280, videoFrameHeight=720, audioCodec=3, audioBitRate=320000, audioSampleRate=48000, audioChannels=2}
01-10 16:19:05.379  3628  3628 D VideoCapabilities: profile = CamcorderProfileProxy{duration=30, quality=4, fileFormat=2, videoCodec=2, videoBitRate=6000000, videoFrameRate=30, videoFrameWidth=720, videoFrameHeight=480, audioCodec=3, audioBitRate=320000, audioSampleRate=48000, audioChannels=2}
[... cut ...]
01-10 16:19:05.557  3628  3756 D CameraStateMachine: Publishing new public camera state CameraState{type=CLOSED, error=null}
01-10 16:19:05.561  3331  3993 I OMX-VENC: Component Deinit
01-10 16:19:05.562  3331  3993 I OMX-VENC: Video encode perflock released
01-10 16:19:05.568  3628  4012 D VideoEncoder: Transitioning encoder internal state: CONFIGURED --> RELEASED
01-10 16:19:05.977  3308  4189 E ResolverController: No valid NAT64 prefix (100, <unspecified>/0)
01-10 16:19:08.002  3308  4191 E ResolverController: No valid NAT64 prefix (100, <unspecified>/0)
```

mrousavy, 2022-01-10:
Maybe try disabling `video` or `photo` or `frameProcessor`?

mrousavy, 2022-01-10:
> How should I do?

What do you mean? Disable `video` or `photo` or `frameProcessor`. Just don't pass them to the `<Camera>` component

mrousavy, 2022-01-10:
Does the flash not fire when taking a photo or when recording (or both)?

mrousavy, 2022-01-10:
HDR is a bit weird - it works when the device supports it, but when the device doesn't support it it is somehow still being reported as "supported".

Angelk90, 2022-01-10:
@mrousavy : It also happens that with the main camera the videos are not recorded.
If I record a video with the secondary camera they are recorded.

mrousavy, 2022-01-10:
> You don't even see the HDR icon.

It's an icon that's missing. In an example app. I have _even_ fixed it right now.

> On android there seems to be a lot of problems.

Thanks, I actually enabled forking on this repository so you can send me PRs fixing those problems anytime :)

> What android device have you tested on?

Huawei P10

>  It also happens that with the main camera the videos are not recorded.

What is the main camera, and what is the second camera?

Angelk90, 2022-01-10:
@mrousavy : Main camera the one you use to take pictures normally (those behind the device)
Secondary room for selfies (those on the device screen).

Angelk90, 2022-01-10:
@mrousavy : 
I pull on the new changes.

- Icon Hdr on see
- The goBack is not working

Angelk90, 2022-01-10:
@mrousavy : I tried on the Tablet Samsung Tab A (2016), to make the camera work I have to disable the video.

ShaheerArshad-nearpeer, 2022-01-12:
[image]
Can any one help me what i am doing wrong ?
Im using react native vision camera
 const devices = useCameraDevices()
  const device = devices.back

Angelk90, 2022-01-12:
@ShaheerArshad-nearpeer : Did you create a new project and then install the library?  Are you testing on an Android emulator or Android physical device?

You should ask @mrousavy.

Angelk90, 2022-01-15:
@mrousavy : Are there any news?

Stukz, 2022-01-24:
**What were you trying to do?**
clone the project: https://github.com/mrousavy/react-native-vision-camera 

try to run the example code on emulator (Pixel 2 XL API 29) & got the same error on the `back` camera

**What happened instead?**

[image]

------

also I have the same error on my physical device **Motorola Moto G20** on both cameras `front` & `back`
[image]

Angelk90, 2022-01-24:
@Stukz : Does the flash work on the g20?

Stukz, 2022-01-24:
> @Stukz : Does the flash work on the g20?

@Angelk90 it doesn't work

Angelk90, 2022-01-24:
@Stukz : What other devices have you tried on?  Both cameras don't work?

Stukz, 2022-01-24:
> @Stukz : What other devices have you tried on? Both cameras don't work?

I only have my g20 as a physical device. yes, both cameras have the same issue. 

it's happening with `video={true}`

Angelk90, 2022-01-24:
@Stukz : I have tested on several Android devices it seems to not work correctly, if you say it does not work on the emulator either it will probably @mrousavy have to be able to test it against a physical device.

Ps
I saw that you are a sponsor what are the advantages?

Stukz, 2022-01-24:
got it, so it's not a specific device, sad to hear that.

> Ps I saw that you are a sponsor what are the advantages?

you can read them on his page. the fact that he's trying to maintain a critical feature like a Camera library on RN it's something to appreciate. I know it's not that much but I hope in the near future to increase my sponsors over the RN OSS maintainers

Angelk90, 2022-01-24:
@Stukz : We hope that this problem on Android will be solved in the future.
From what I understand the problem appears to be camerax, but I don't understand why.

mrousavy, 2022-01-25:
Hey! So the problem here is the core difference between iOS Cameras (AVFoundation) and Android Cameras (CameraX).

On iOS, the concept is similar to how I expose it in VisionCamera, you have a list of devices, each has a list of formats. With CameraX, all of this is a bit abstracted away and you only have a list of devices, but the concept of "formats" does not really exist.

On Android, some Camera devices do not support `photo` and `video` at the same time. I am not sure why this is the case, but on iOS this always works.

Also, apparently on CameraX there is no way to determine whether a camera supports `video` and `photo` at the same time _before mounting the Camera_, which is just dumb. I can only use guesses to find out if a device in the devices list supports `video` and `photo`.

I have a contact to Google and will ask him a few questions, maybe I can come up with an alternative API for VisionCamera that solves this issue. Sorry for the complications, but this is a tough one!

mrousavy, 2022-01-25:
And thank you for the sponsorship @Stukz, means a lot to me! 🙏

Stukz, 2022-01-26:
Hey Marc thank you for the detailed explanation of what's going on in Android, let me know if you have any progress on it so I can give you some feedback about it

Stukz, 2022-03-03:
hey @mrousavy hope you are doing well, I'm curious if you have any update about this issue?

mrousavy, 2022-03-04:
Hey Nicolas, I didn't investigate this any further for now since I didn't have the time to do so. I am thinking about a new API that could potentially solve this issue!

NgocLe1001, 2022-03-14:
I have same issue. Any one has solution please :(

NgocLe1001, 2022-03-14:
I removed all related `frameRateIncluded` then it works. It's fix temporarily

Angelk90, 2022-03-15:
@ozgurbayram, @NgocLe1001  : On five android devices I tested on, it didn't work properly.
Seems like some problem with cameraX, when it says @mrousavy.

mrousavy, 2022-03-16:
CameraX is really not as good as they make it sound... :/

Angelk90, 2022-03-16:
@mrousavy : So for the moment on android it doesn't work properly, right?

nimran, 2022-06-18:
@mrousavy  unable to capture image in **Android**. I dont see any logs either, it works in iOS when i click,it captures.

mikamations, 2022-11-21:
This is still an issue in 2.15.2. Most Android devices don't work unless video is disabled. Anyone find any better workarounds?

mrousavy, 2022-12-13:
Hey all! CameraX is just too limiting. It's not based on Camera Device Formats as AVFoundation is, so I need to do a little rewriting for the Android part to make everything work as flexible and powerful as on iOS. 

For this part specifically, I need to:

1. Rewrite the native Android part from CameraX to Camera2
2. Create a new declarative API ontop of the native iOS and Android parts to make this whole Camera Device + Format selection way easier (_describe_ your target FPS, HDR, resolution, etc and it will pick everything automatically)

However since this is a really big project, I can't really work on this in my free time (I have a company to run). I have created a discussion for this, and want to put it all into VisionCamera V3, if you or your company is interested in this new version and wants to help support/fund the development of that, please reach out to me via Twitter or Email. 

Here's the technical discussion: https://github.com/mrousavy/react-native-vision-camera/issues/1376

closed 2023-09-01 by mrousavy