margelo/react-native-vision-camera #730: 🐛

HartWoom, 2022-01-10:
### What were you trying to do?

I'm trying to integrate this community plugin : https://github.com/rodgomesc/vision-camera-code-scanner
But I'm unable to make it work on iOS. It works great on Android though.
I have tried other community plugin but the same error appear. In fact event a frame processor function with no code in it makes the error appear.
I've also tried different version react-native-vision-camera, since some changes has been made since the last version of vision-camera-code-scanner.

### Reproduceable Code

```tsx
const ScanBarCodeScreen = (props: NavigationProps) => {
  const [hasPermission, setHasPermission] = useState(false)
  const devices = useCameraDevices()
  const device = devices.back

  const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    console.log('hello')
  }, [])

  const isFocused = useIsFocused()

  useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission()
      setHasPermission(status === 'authorized')
    })()
  }, [])

  return (
    <>
      <StatusBar backgroundColor={Theme.Color.White} barStyle="dark-content" />
      <View
        style={{
          flex: 1,
          flexDirection: 'column',
          backgroundColor: 'black'
        }}
      >
        <SafeAreaView
          style={{
            flex: 1,
            justifyContent: 'space-between',
            alignItems: 'center'
          }}
        >
          {device != null && hasPermission && (
            <Camera
              style={StyleSheet.absoluteFill}
              device={device}
              isActive={isFocused}
              frameProcessor={frameProcessor}
              frameProcessorFps={1}
            />
          )}
        </SafeAreaView>
      </View>
    </>
  )
}
```

### What happened instead?

As soon as I arrive on my screen, and after giving permission to use the device camera, I get the error `Frame Processor threw an error: Value is undefined, expected an Object` at every refresh.

The app does not crash but I obviously cannot use the frame processor.

### Relevant log output

```shell
FrameProcessorBindings: Creating Runtime Manager...
FrameProcessorBindings: Runtime Manager created!
FrameProcessorBindings: Installing Frame Processor plugins...
FrameProcessorBindings: Installing Frame Processor plugin "__scanCodes"...
FrameProcessorBindings: Installing Frame Processor plugin "__labelImage"...
FrameProcessorBindings: Frame Processor plugins installed!
FrameProcessorBindings: Installing Frame Processor Bindings for Bridge...
FrameProcessorBindings: Installing global functions...
FrameProcessorBindings: Finished installing bindings.
[javascript] Running "TitrePrepayesApp" with {"rootTag":1,"initialProps":{}}
[native] [GESTURE HANDLER] Initialize gesture handler for root view <RCTRootContentView: 0x104e19640; reactTag: 1; frame = (0 0; 768 1024); gestureRecognizers = <NSArray: 0x282a4a220>; layer = <CALayer: 0x28243d5e0>>
[MC] System group container for systemgroup.com.apple.configurationprofiles path is /private/var/containers/Shared/SystemGroup/systemgroup.com.apple.configurationprofiles
[MC] Reading from public effective user settings.
[Common] _BSMachError: port b003; (os/kern) invalid capability (0x14) "Unable to insert COPY_SEND"
[javascript] 'Frameporcccc', [Function]
FrameProcessorBindings: Setting new frame processor...
FrameProcessorBindings: Adapting Shareable value from function (conversion to worklet)...
FrameProcessorBindings: Successfully created worklet!
FrameProcessorBindings: Converting worklet to Objective-C callback...
FrameProcessorBindings: Frame processor set!
[javascript] Frame Processor threw an error: Value is undefined, expected an Object
[javascript] Frame Processor threw an error: Value is undefined, expected an Object
[javascript] Frame Processor threw an error: Value is undefined, expected an Object
```

### Device

iPad mini 2 on iOS 12.5.5

### VisionCamera Version

2.11.2 ; 2.11.1 ; 2.9.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-01-10:
Hi! I see, does this happen on JSC or Hermes? cc @rodgomesc

HartWoom, 2022-01-10:
Hi ! I'm sorry but I have no clue of what either of those things are :no_mouth:

mrousavy, 2022-01-10:
The JS engine. You can switch between JSC (the default) and Hermes. Assuming you're using RN 0.64 or above

HartWoom, 2022-01-10:
I'm on JSC, I've just checked if 'enableHermes' was on false. And I'm on RN 0.64.2.

mrousavy, 2022-01-10:
Try if enabling Hermes fixes the issue for you

HartWoom, 2022-01-10:
Enabling Hermes did not fix it :/ Still have a `Value is undefined, expected an Object` error. It does something weird though, when I try to scan a barcode, xcode then opens me the Debug Navigator to show me memory addresses I think ? So the frame processor does seem to detect a Barcode.

khudratov, 2022-01-14:
Hi. your babel.config.js like this?
module.exports = {
  presets: ['module:metro-react-native-babel-preset'],
  "plugins": [
    [
      "react-native-reanimated/plugin",
      {
        "globals": ["__scanCodes"],
      },
    ],
  ],
};

HartWoom, 2022-01-14:
Hi @khudratov, yup exactly the same

khudratov, 2022-01-14:
did you import:  import 'react-native-reanimated'; in root index.js

HartWoom, 2022-01-14:
Nope I did not ! I just tried again by adding the import, unfortunately the same error keeps appearing

mrousavy, 2022-01-14:
Does https://github.com/mrousavy/react-native-vision-camera/pull/698 fix the issue for you? Try rebuilding everything and cleaning cache (especially metro cache)

mrousavy, 2022-01-14:
Which REA version are you on?

khudratov, 2022-01-14:
can you share your repo?

Jamal-Nazzal, 2022-01-15:
same problem here: Frame Processor threw an error: Value is undefined, expected an Object.

`    const frameProcessor = useFrameProcessor((frame) => {
        'worklet';
        console.log(frame);
    }, []);`
`            <Camera
                ref={camera}
                style={StyleSheet.absoluteFill}
                device={device}
                isActive={true}
                format={format}
                photo={true}
                frameProcessor={frameProcessor}
            />`

hermes is on, reanimated and gesture-handler are imported.

Jamal-Nazzal, 2022-01-15:
is the error coming from react-native-vision-camera\android\src\main\cpp\FrameProcessorRuntimeManager.cpp ?

`      cameraView->cthis()->setFrameProcessor([this, &rt, function](jni::alias_ref<JImageProxy::javaobject> frame) {
          try {
            // create HostObject which holds the Frame (JImageProxy)
            auto hostObject = std::make_shared<FrameHostObject>(frame);
            function->callWithThis(rt, *function, jsi::Object::createFromHostObject(rt, hostObject));
          } catch (jsi::JSError& jsError) {
            auto message = "Frame Processor threw an error: " + jsError.getMessage();
            __android_log_write(ANDROID_LOG_ERROR, TAG, message.c_str());
            this->logErrorToJS(message);
          }
      });`
why is Camera unable to pass a frame to the frameProcessor hook? is it somewhere in the setAnalyzer method? for some reason it's not passing the data stream coming from the camera to the frameProcessor callback!

Jamal-Nazzal, 2022-01-15:
ok it's working now, I did the following:
1.  npm start -- --reset-cache
2. cd android
3. ./gradlew clean

that did it. I previously tried clean without resetting the cache but that didn't help at all. npm start -- --reset-cache as suggested by mrousavy seems to solve it

closed 2022-01-16 by mrousavy

HartWoom, 2022-01-19:
Hi, sorry for the long delay, I tried applying #698 and cleaning the metro  cache but it didn't workout. I also cleaned the ios build folder, but the same error keep appearing.
I'm letting this on the side for now.

mayfairr, 2022-02-04:
Hi,

I'm experiencing the same issue as mentioned above. I have Hermes enabled with both Gesture-Handler and React-Reanimated imported in `index.js`.  This is a bare React Native Project ` "react-native": "0.67.2"`

Any suggestions?

mrousavy, 2022-02-04:
Yes, I need that function. In the future I have an alternative to Reanimated ready, so stay tuned

gh-pro, 2022-06-18:
hi, anyone resolve this issue ?

Vladislava9009, 2022-06-20:
I have the same issue

davidjburgos, 2022-06-27:
Any comment on this? I have the same issue. 

I noticed that I dont have import 'react-native-reanimated'; in index.js because if I added then I get the error:

Error: Reanimated 2 failed to create a worklet, maybe you forgot to add Reanimated's babel plugin?

Any help, please?

mrousavy, 2022-06-30:
Did you install Reanimated's babel plugin?

saidybarry, 2022-07-02:
Guys can anyone help me out am I having this error below,

 ERROR  Frame Processor threw an error: scanBarcodes is not a function. (In 'scanBarcodes(frame,[BarcodeFormat.QR_CODE],{checkInverted:true})', 'scanBarcodes' is undefined)

heckid18, 2022-07-28:
I have had a similar issue and fixed it by downgrading react-native-reanimated to 2.4.1.

jimshidbachu, 2022-07-29:
> enabling Hermes

enabled the Herms, but it shows like this always

What went wrong:
Execution failed for task ':app:mergeDebugNativeLibs'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.MergeNativeLibsTask$MergeNativeLibsTaskWorkAction
   > 2 files found with path 'lib/arm64-v8a/libfolly_runtime.so' from inputs:
      - /home/jimshid/codesap/Projects/waseet/node_modules/react-native-vision-camera/android/build/intermediates/library_jni/debug/jni/arm64-v8a/libfolly_runtime.so
      - /home/jimshid/.gradle/caches/transforms-3/ad1868fce16913a195c4e3d35d5c12bb/transformed/jetified-react-native-0.69.1-debug/jni/arm64-v8a/libfolly_runtime.so
     If you are using jniLibs and CMake IMPORTED targets, see
     https://developer.android.com/r/tools/jniLibs-vs-imported-targets

thelastwizardOC, 2022-08-11:
> Guys can anyone help me out am I having this error below,
> 
> ERROR Frame Processor threw an error: scanBarcodes is not a function. (In 'scanBarcodes(frame,[BarcodeFormat.QR_CODE],{checkInverted:true})', 'scanBarcodes' is undefined)

@saidybarry Hello, have you found any solution for this problem?

heckid18, 2022-08-11:
Maybe you have not installed the package or did not import it. Also make sure you have correctly configured babel.config.js.

Fritzzzz1, 2022-08-23:
I had the same problem (
ERROR Frame Processor threw an error: scanBarcodes is not a function. (In 'scanBarcodes(frame,[BarcodeFormat.QR_CODE],{checkInverted:true})', 'scanBarcodes' is undefined)
)
Probably cause by linter or formatter - that does not accept imports from '.'
Fix : in node-modules/vision-camera-code-scanner/src/hook.tsx changed line 5 to :

import { Barcode, BarcodeFormat, CodeScannerOptions, scanBarcodes } from './index'
instead of '.'

Also patched the package.

saidybarry, 2022-08-23:
> > Guys can anyone help me out am I having this error below,
> > ERROR Frame Processor threw an error: scanBarcodes is not a function. (In 'scanBarcodes(frame,[BarcodeFormat.QR_CODE],{checkInverted:true})', 'scanBarcodes' is undefined)
> 
> @saidybarry Hello, have you found any solution for this problem?

@thelastwizardOC nope I wasn't able to find a solution, I did end up switching to a different package.

jimshidbachu, 2022-08-24:
> @thelastwizardOC nope I wasn't able to find a solution, I did end up switching to a different package.
 
to which package you switched?

saidybarry, 2022-08-24:
> to which package you switched?

@thelastwizardOC Mhmm actually i was trying to use it in relation to another package to implement QR-code scanning, so after getting stuck with react native vision camera i decided to up for "react-native-qrcode-scanner".

jyotishman, 2022-10-17:
follow the answer in this link to solve the issue-
[](https://stackoverflow.com/questions/73817506/frame-processor-threw-an-error-value-is-undefined-expected-an-object)