margelo/react-native-vision-camera #733: FAILURE: Build failed with an exception. (Android) 🐛

siddharth-kt, 2022-01-11:
### What were you trying to do?

Trying to install react-native-vision-camera.

### What happened instead?

The build failed on ANDROID.

One thing i want to mention is that there is a similar issue https://github.com/mrousavy/react-native-vision-camera/issues/515, BUT the workarounds mentioned over there can not solve this issue because this issue log also mentions about androidx.camera:camera-video:1.1.0-alpha12.
And even the oldest version on androidx.camera:camera-video is 1.1.0-alpha10 which also give similar error.

I am using bare workflow but i am also using few libs from expo (expo-modules) due to which I can not upgrade to compileSdkVersion 31.

### Relevant log output

```shell
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:checkDebugAarMetadata'.
> Multiple task action failures occurred:
   > A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
      > The minCompileSdk (31) specified in a
        dependency's AAR metadata (META-INF/com/android/build/gradle/aar-metadata.properties)
        is greater than this module's compileSdkVersion (android-30).
        Dependency: androidx.camera:camera-view:1.0.0-alpha32.
        AAR metadata file: C:\Users\sidty\.gradle\caches\transforms-2\files-2.1\5047891a4162a71f93ff85ab223dcb21\jetified-camera-view-1.0.0-alpha32\META-INF\com\android\build\gradle\aar-metadata.properties.
   > A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
      > The minCompileSdk (31) specified in a
        dependency's AAR metadata (META-INF/com/android/build/gradle/aar-metadata.properties)
        is greater than this module's compileSdkVersion (android-30).
        Dependency: androidx.camera:camera-extensions:1.0.0-alpha32.
        AAR metadata file: C:\Users\sidty\.gradle\caches\transforms-2\files-2.1\efcf661c50e10c53fcd3269b9f4248bb\jetified-camera-extensions-1.0.0-alpha32\META-INF\com\android\build\gradle\aar-metadata.properties.
   > A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
      > The minCompileSdk (31) specified in a
        dependency's AAR metadata (META-INF/com/android/build/gradle/aar-metadata.properties)
        is greater than this module's compileSdkVersion (android-30).
        Dependency: androidx.camera:camera-camera2:1.1.0-alpha12.
        AAR metadata file: C:\Users\sidty\.gradle\caches\transforms-2\files-2.1\bbfe3ad27c727c2aefe1d2bd8d71de65\jetified-camera-camera2-1.1.0-alpha12\META-INF\com\android\build\gradle\aar-metadata.properties.
   > A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
      > The minCompileSdk (31) specified in a
        dependency's AAR metadata (META-INF/com/android/build/gradle/aar-metadata.properties)
        is greater than this module's compileSdkVersion (android-30).
        Dependency: androidx.camera:camera-lifecycle:1.1.0-alpha12.
        AAR metadata file: C:\Users\sidty\.gradle\caches\transforms-2\files-2.1\5c290337d3df02444950d468325cb7b9\jetified-camera-lifecycle-1.1.0-alpha12\META-INF\com\android\build\gradle\aar-metadata.properties.
   > A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
      > The minCompileSdk (31) specified in a
        dependency's AAR metadata (META-INF/com/android/build/gradle/aar-metadata.properties)
        is greater than this module's compileSdkVersion (android-30).
        Dependency: androidx.camera:camera-video:1.1.0-alpha12.
        AAR metadata file: C:\Users\sidty\.gradle\caches\transforms-2\files-2.1\517e3ba9cc7450c5de251bea701a08be\jetified-camera-video-1.1.0-alpha12\META-INF\com\android\build\gradle\aar-metadata.properties.
   > A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
      > The minCompileSdk (31) specified in a
        dependency's AAR metadata (META-INF/com/android/build/gradle/aar-metadata.properties)
        is greater than this module's compileSdkVersion (android-30).
        Dependency: androidx.camera:camera-core:1.1.0-alpha12.
        AAR metadata file: C:\Users\sidty\.gradle\caches\transforms-2\files-2.1\050175e652e788ef96f1c4a65ee0556a\jetified-camera-core-1.1.0-alpha12\META-INF\com\android\build\gradle\aar-metadata.properties.
```

### Device

Vivo v15 pro

### VisionCamera Version

2.11.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

siddharth-kt, 2022-01-12:
@mrousavy can you please tell any workaround.

mrousavy, 2022-01-13:
Yes, your error message already tells you that you need to change `compileSdkVersion` version to `31`

closed 2022-01-13 by mrousavy

siddharth-kt, 2022-01-13:
So there can't be any solution than to set compileSdkVersion to 31 ?

mrousavy, 2022-01-13:
No, it is required by the native Android Camera library (CameraX).

siddharth-kt, 2022-01-13:
ok, i got it.

Thanks