margelo/react-native-vision-camera #740: 🐛 Device undefined

BugraTuncer, 2022-01-13:
### What were you trying to do?

I tried troubleshooting steps. build.gradle and androidmanifest.xml same as example.

And also logcat shows : 

2-01-13 22:08:47.541 3811-4075/com.temp W/System.err: java.lang.NoSuchMethodError: No static method getInstanceAsync(Landroid/content/Context;Landroidx/camera/core/CameraProvider;)Lcom/google/common/util/concurrent/ListenableFuture; in class Landroidx/camera/extensions/ExtensionsManager; or its super classes (declaration of 'androidx.camera.extensions.ExtensionsManager' appears in /data/app/~~ug-pT-hZ3WmIwtf3FkQWeg==/com.temp-5EZI6bZq5H2-dMB8IR4JnA==/base.apk)

### Reproduceable Code

```tsx
const camera = React.useRef<any>(null);

  const isFocussed = useIsFocused();
  const isForeground = useIsForeground();
  const isActive = isFocussed && isForeground;

  const [cameraPosition, setCameraPosition] = React.useState<'front' | 'back'>(
    'back',
  );
  const devices = useCameraDevices();
  const device = devices[cameraPosition];

  const handleSubmit = () => {
    console.log(device);
  };

  const onError = React.useCallback((error: CameraRuntimeError) => {
    console.error(error);
  }, []);

  if (device != null) {
    console.log(
      `Re-rendering camera page with ${
        isActive ? 'active' : 'inactive'
      } camera. ` + `Device: "${device.name}"`,
    );
  } else {
    console.log('re-rendering camera page without active camera');
  }

  return (
    <SafeAreaView style={{flex: 1}}>
      {device != null && (
        <Camera
          device={device}
          ref={camera}
          isActive
          video={true}
          audio={true}
          photo={true}
          onError={onError}
        />
      )}
      
    </SafeAreaView>
  );
```

### What happened instead?

It always comes device undefined. I define the permissions, the microphone and camera are authorized, then I redirect the page and I get the error below.

### Relevant log output

```shell
2022-01-13 21:44:31.501 2576-2623/com.temp W/ReactNativeJS: Possible Unhandled Promise Rejection (id: 0):
    unknown/unknown: [unknown/unknown] No static method getInstanceAsync(Landroid/content/Context;Landroidx/camera/core/CameraProvider;)Lcom/google/common/util/concurrent/ListenableFuture; in class Landroidx/camera/extensions/ExtensionsManager; or its super classes (declaration of 'androidx.camera.extensions.ExtensionsManager' appears in /data/app/~~FXHKVXJMs-IDkwiM2eg40g==/com.temp -nGCO2mBqZvehaHv8rq5wEQ==/base.apk)
    construct@[native code]
    _construct@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:22959:28
    Wrapper@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:22912:64
    construct@[native code]
    _createSuperInternal@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:117041:322
    CameraError@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:117054:26
    construct@[native code]
    _createSuperInternal@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp&modulesOnly=false&runModule=true:117041:322
    CameraRuntimeError@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp&modulesOnly=false&runModule=true:117109:27
    tryParseNativeCameraError@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp&modulesOnly=false&runModule=true:117132:38
    getAvailableCameraDevices$@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp&modulesOnly=false&runModule=true:116898:102
    tryCatch@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:24305:23
    invoke@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp&modulesOnly=false&runModule=true:24475:32
    tryCatch@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp&modulesOnly=false&runModule=true:24305:23
    invoke@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:24377:30
    http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:24389:21
    tryCallOne@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:28982:16
    http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:29083:27
    http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:29674:26
    _callTimer@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:29570:17
    _callReactNativeMicrotasksPass@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:29609:17
    callReactNativeMicrotasks@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:29819:44
    __callReactNativeMicrotasks@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:5432:46
    http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:5210:45
    __guard@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:5415:15
    flushedQueue@http://10.0.2.2:8081/index.bundle?platform=android&dev=true&minify=false&app=com.temp &modulesOnly=false&runModule=true:5209:21
2022-01-13 21:44:31.501 2576-2623/com.temp W/ReactNativeJS: flushedQueue@[native code]
    invokeCallbackAndReturnFlushedQueue@[native code]
```

### Device

Pixel 2 Api 30

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-01-14:
That's very interesting, it looks like an issue with the native CameraX library?

batical, 2022-03-06:
same issue here. Any tips ?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy