margelo/react-native-vision-camera #742: ✨ startRecording callback/promise

Hirbod, 2022-01-14:
### What feature or enhancement are you suggesting?

Would be lovely to have a callback or / be able to use `await startRecording()` in order to make sure that recording indeed did start.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

No workaround currently

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2022-01-14:
https://github.com/mrousavy/react-native-vision-camera/blob/00fc39891a5cd6c68121fd5cd5f6ede90c9eb333/ios/CameraView%2BRecordVideo.swift#L60-L63

Hirbod, 2022-01-14:
Ok, I see! Gonna live with that for now :)

mrousavy, 2023-11-21:
Started working on something here: https://github.com/mrousavy/react-native-vision-camera/pull/2194

as of now this doesn't really work, but we'll see what we can do

mrousavy, 2024-01-15:
Closing because this is being tracked in https://github.com/mrousavy/react-native-vision-camera/issues/1319