margelo/react-native-vision-camera #746: 🐛 Android- frame processor alway return wrong width/height of frame

vanhungoz, 2022-01-17:
### What were you trying to do?

i try to use frame to recognize face

### Reproduceable Code

```tsx
const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    console.log('frame.width', frame.width);
    console.log('frame.height', frame.height);
  }, []);
```

### What happened instead?

frame data alway wrong.
camera's orientation is portrait
device's orientation is portrait.

### Relevant log output

```shell
LOG  frame.width 1280
LOG  frame.height 720
```

### Device

pixel 3L. Android 12

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

vanhungoz, 2022-01-17:
[image]
frame preview is showing correctly. But frame data may be wrong
it is only happening on Android. iOS is working correctly

vanhungoz, 2022-01-17:
it failed  on both front and back camera.

mrousavy, 2022-01-17:
Okay, maybe this commit broke something: https://github.com/mrousavy/react-native-vision-camera/commit/48da1819fc0fed894cd5c3ca994f9a5c93ed757b

vanhungoz, 2022-01-17:
@mrousavy  thank you for your quick response.

i have checked `"react-native-vision-camera": "2.11.1",`
i still got the same error

vanhungoz, 2022-01-17:
i checked release history, orientation is added from version 2.11.2

mrousavy, 2022-01-17:
I mean technically it would be correct. The width is always the physical sensor width. But I agree that we can use the logical term width and height instead, just like on iOS

mrousavy, 2022-01-17:
I just wonder where it goes wrong, here we set the target rotation: https://github.com/mrousavy/react-native-vision-camera/blob/28fc6a68a5744efc85b532a338e2ab1bc8fa45fe/android/src/main/java/com/mrousavy/camera/CameraView.kt#L396-L397

Could you maybe check here what `image.width` is? https://github.com/mrousavy/react-native-vision-camera/blob/28fc6a68a5744efc85b532a338e2ab1bc8fa45fe/android/src/main/java/com/mrousavy/camera/CameraView.kt#L477-L493

vanhungoz, 2022-01-17:
ok, thank you, i'll check it and tell you

vanhungoz, 2022-01-17:
@mrousavy  it alway log `1280` as same as frame.width

bkidai, 2022-01-17:
@vanhungoz @mrousavy isn't the wrong `with`/`height` it's because on Android the frame it's rotated with 90° degree (i'm having the same _issue_) even before the last #715

vanhungoz, 2022-01-18:
@bkidai  have you tested with old version. Which version could we use to check frame processor with correct frame?

vanhungoz, 2022-01-19:
> @vanhungoz @mrousavy isn't the wrong `with`/`height` It's because on Android the frame it's rotated by 90° degree (i'm having the same _issue_) even before the last commit #715
> 
> Btw, i have the same output when i use `orientation="portrait"` `console.log(frame.toString());` `1280 x 720 Frame // output`

@bkidai  have you found any workaround?

bkidai, 2022-01-19:
@vanhungoz i rotate the image based on the `ImageProxy.imageInfo.rotationDegrees` natively in Android before using it

bkidai, 2022-01-19:
> @bkidai have you tested with older version. Which version could we use to check frame processor with correct frame?
I had the same issue on the older versions too

vanhungoz, 2022-01-22:
@mrousavy  do you have any suggest to find out this bug?

vanhungoz, 2022-01-24:
> 

`matrix.setRotate((360- image.imageInfo.rotationDegrees).toFloat())
            val rotatedBitmap = Bitmap.createBitmap(bitmap, 0, 0, bitmap.width, bitmap.height,matrix,true)`
            
            @bkidai could you share your code.
            i'm rotating the image by this function

bkidai, 2022-01-24:
@vanhungoz here's my code
```
matrix.postRotate(rotationDegrees.toFloat())
val rotatedBitmap = Bitmap.createBitmap(bitmap, 0, 0, image.width, image.height, matrix, true)
```

vanhungoz, 2022-01-24:
thank you, the same as me

xulihang, 2022-02-11:
If we rotated the image on the Android side, the `frame.width` is still wrong on the JavaScript side. Maybe frame should have a `rotationDegrees` prop.

xulihang, 2022-02-11:
In order to show barcode overlays, I rotated the image on the Android side using the yuv2bitmap converting code from here: https://blog.minhazav.dev/how-to-convert-yuv-420-sp-android.media.Image-to-Bitmap-or-jpeg/#pure-java-approach

I then use Dimensions to infer whether I should switch frame width and frame height:

```js
if (frameHeight>frameWidth && Dimensions.get('window').width>Dimensions.get('window').height){
      console.log("Has rotation");
      viewBox = "0 0 "+frameHeight+" "+frameWidth;
    }else {
      viewBox = "0 0 "+frameWidth+" "+frameHeight;
    }    
```

Full example: https://github.com/xulihang/vision-camera-dynamsoft-barcode-reader/tree/main/example

mrousavy, 2022-02-11:
Maybe this fixes the problem: https://github.com/mrousavy/react-native-vision-camera/pull/816

xulihang, 2022-02-11:
The new [setOutputImageRotationEnabled](https://developer.android.com/jetpack/androidx/releases/camera#1.1.0-alpha11) API used in the pr seems the correct way.

yachaka, 2022-02-22:
I had the same issue
So I did

```js
// If frame is inverted  
// This happens on Pixel 2, width and height are inverted
        if (screenWidth < screenHeight && frameWidth > frameHeight
            || ( screenWidth > screenHeight && frameWidth < frameHeight)) {
            const tmp = frameWidth;
            frameWidth = frameHeight;
            frameHeight = tmp;
        }
```

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy