margelo/react-native-vision-camera #747: 🐛 takePhoto does not return on example/ app, in specific emulator

savv, 2022-01-17:
### What were you trying to do?

I wanted to take a picture :)

### Reproduceable Code

```tsx
https://github.com/mrousavy/react-native-vision-camera/tree/main/example using
this emulator:
Device: "Pixel XL"
Image: "Android API 31 (Google APIs) x86_64"
```

### What happened instead?

```
 DEBUG  state: ACTIVE
 DEBUG  state: END
 LOG  Taking photo...
```

and it hangs there forever.

### Device

Emulator for "Pixel XL" using Image "Android API 31 (Google APIs) x86_64"

### VisionCamera Version

HEAD

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-01-17:
This is indeed very weird. Either it's falling back to Snapshot (not logged, so I guess not?), or it's a CameraX bug..

savv, 2022-01-19:
FYI, the logcat logs. This one is from Pixel 4 XL (instead of just Pixel XL). Same image.

```
01-19 11:45:25.046 22811 22902 D ReactNativeJS: state: BEGAN
01-19 11:45:25.085 22811 22830 D EGL_emulation: app_time_stats: avg=3729.76ms min=1.59ms max=92872.04ms count=25
01-19 11:45:25.088 22811 22902 D ReactNativeJS: state: ACTIVE
01-19 11:45:25.089 22811 22902 D ReactNativeJS: state: END
01-19 11:45:25.090 22811 22902 I ReactNativeJS: Taking photo...
01-19 11:45:25.098 22811 22894 I CameraView: takePhoto() called
01-19 11:45:25.100 22811 22898 D UseCaseAttachState: Active and attached use case: [androidx.camera.video.VideoCapture-29ae22e7-ebba-4596-9fc0-e037d144ffd846650572, androidx.camera.core.ImageCapture-4f489441-14b4-4b53-86a8-11dfdc3fa3e962124821, androidx.camera.core.Preview-c760ae44-8138-4668-b6c5-471b5d43bd2185057279] for camera: 0
01-19 11:45:25.101 22811 22893 D CameraView: Taking picture...
01-19 11:45:25.102 22811 22898 D CaptureSession: Attempting to submit CaptureRequest after setting
01-19 11:45:25.102 22811 22898 D CaptureSession: Issuing request for session.
01-19 11:45:25.104 22811 22894 D CameraView: Creating temp file...
01-19 11:45:25.106 22811 22811 D CameraOrientationUtil: getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
01-19 11:45:25.106 22811 22811 D ImageCapture: Send image capture request [current, pending] = [0, 1]
01-19 11:45:25.109 22811 22943 D ImageCapture: issueTakePicture
01-19 11:45:25.109 22811 22897 D Camera2CameraImpl: {Camera@fc88e22[id=0]} Issue capture request
01-19 11:45:25.110 22811 22897 D CaptureSession: Issuing capture request.
01-19 11:45:25.132   363   419 D goldfish-address-space: claimShared: Ask to claim region [0x3e3ed7000 0x3e3f7c000]
01-19 11:45:25.148   363   419 D goldfish-address-space: claimShared: Ask to claim region [0x3e01c2000 0x3e0384000]
01-19 11:45:25.166 21795 22944 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 22944 (EmulatedFakeCam), pid 21795 (provider@2.4-se)
01-19 11:45:25.206 22947 22947 I crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstoneProto
01-19 11:45:25.208   253   253 I tombstoned: received crash request for pid 22944
01-19 11:45:25.209 22947 22947 I crash_dump64: performing dump of process 21795 (target tid = 22944)
01-19 11:45:25.224 22947 22947 E DEBUG   : failed to read /proc/uptime: Permission denied
01-19 11:45:25.300     0     0 I logd    : logdr: UID=1047 GID=1005 PID=22947 n tail=0 logMask=8 pid=21795 start=0ns deadline=0ns
01-19 11:45:25.309     0     0 I logd    : logdr: UID=1047 GID=1005 PID=22947 n tail=0 logMask=1 pid=21795 start=0ns deadline=0ns
01-19 11:45:25.349 22947 22947 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
01-19 11:45:25.349 22947 22947 F DEBUG   : Build fingerprint: 'google/sdk_gphone64_x86_64/emulator64_x86_64_arm64:12/SE1A.211012.001/7818354:userdebug/dev-keys'
01-19 11:45:25.349 22947 22947 F DEBUG   : Revision: '0'
01-19 11:45:25.349 22947 22947 F DEBUG   : ABI: 'x86_64'
01-19 11:45:25.349 22947 22947 F DEBUG   : Timestamp: 2022-01-19 11:45:25.219739548+0100
01-19 11:45:25.349 22947 22947 F DEBUG   : Process uptime: 0s
01-19 11:45:25.349 22947 22947 F DEBUG   : Cmdline: /vendor/bin/hw/android.hardware.camera.provider@2.4-service_64
01-19 11:45:25.349 22947 22947 F DEBUG   : pid: 21795, tid: 22944, name: EmulatedFakeCam  >>> /vendor/bin/hw/android.hardware.camera.provider@2.4-service_64 <<<
01-19 11:45:25.349 22947 22947 F DEBUG   : uid: 1047
01-19 11:45:25.349 22947 22947 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
01-19 11:45:25.349 22947 22947 F DEBUG   : Cause: null pointer dereference
01-19 11:45:25.349 22947 22947 F DEBUG   :     rax 0000000000000000  rbx 00007ec443971b10  rcx 0000000000000001  rdx 00007ec443971cf0
01-19 11:45:25.349 22947 22947 F DEBUG   :     r8  00007ec6e7da439f  r9  00007ec44397184f  r10 0000000000000000  r11 00007ec6e7edb4d8
01-19 11:45:25.349 22947 22947 F DEBUG   :     r12 00007ec6e782b500  r13 000000000012c000  r14 00007ec44c46a220  r15 0000000000000050
01-19 11:45:25.349 22947 22947 F DEBUG   :     rdi 00007ec443971b10  rsi 00000000000059a0
01-19 11:45:25.349 22947 22947 F DEBUG   :     rbp 00007ec443971b10  rsp 00007ec443971a68  rip 0000000000000000
01-19 11:45:25.349 22947 22947 F DEBUG   : backtrace:
01-19 11:45:25.349 22947 22947 F DEBUG   :       #00 pc 0000000000000000  <unknown>
01-19 11:45:25.349 22947 22947 F DEBUG   :       #01 pc 0000000000062b40  /vendor/lib64/hw/camera.ranchu.so (android::NV21JpegCompressor::NV21JpegCompressor()+112) (BuildId: c55923ff6fa374f3e8a4a7c2d4d1ec5a)
01-19 11:45:25.349 22947 22947 F DEBUG   :       #02 pc 00000000000000f0  <unknown>
01-19 11:45:25.349 22947 22947 F DEBUG   :       #03 pc 0000000000089ca3  /vendor/lib64/hw/camera.ranchu.so (android::createThumbnail(unsigned char const*, int, int, int, int, int, _ExifData*)+675) (BuildId: c55923ff6fa374f3e8a4a7c2d4d1ec5a)
01-19 11:45:25.349 22947 22947 F DEBUG   :       #04 pc 0000000000000140  <unknown>
01-19 11:45:25.349 22947 22947 F DEBUG   :       #05 pc 000000000006f0d9  /vendor/lib64/hw/camera.ranchu.so (android::JpegCompressor::compress()+441) (BuildId: c55923ff6fa374f3e8a4a7c2d4d1ec5a)
01-19 11:45:25.354   548   888 W NativeCrashListener: Couldn't find ProcessRecord for pid 21795
01-19 11:45:25.355   253   253 E tombstoned: Tombstone written to: tombstone_02
[... cut ...] _tracing ANDROID_EMU_gles_max_version_3_0
01-19 11:45:26.288   363   419 D goldfish-address-space: claimShared: Ask to claim region [0x3e0384000 0x3e0546000]
01-19 11:45:26.525   363   419 D goldfish-address-space: claimShared: Ask to claim region [0x3e0546000 0x3e0708000]
01-19 11:45:26.579   363   419 D goldfish-address-space: claimShared: Ask to claim region [0x3e0708000 0x3e08ca000]
01-19 11:45:27.118 22811 22830 D EGL_emulation: app_time_stats: avg=3.50ms min=1.42ms max=15.45ms count=47
```

mrousavy, 2022-01-19:
Oh so it's actually crashing the app?

savv, 2022-01-19:
Actually, it’s not. It still works just hangs after takePhoto

savv, 2022-01-20:
This seems to be an issue with the emulator

https://github.com/android/camera-samples/issues/432#issuecomment-952121904

https://github.com/android/camera-samples/issues/430

https://github.com/android/camera-samples/issues/428

OvenCroc, 2022-05-05:
have you solved this problem,  the newest version camera-core:1.1.0-beta03 still hang when i take photo

tsal, 2023-01-02:
> have you solved this problem, the newest version camera-core:1.1.0-beta03 still hang when i take photo

The issue is with the emulator device itself. Shut down the emulator device, then edit the settings for the device in the device manager. Click Show Advanced Settings then ensure your Cameras are set to "Emulated" instead of "VirtualScene".

chinnuu05, 2024-07-15:
> The issue is with the emulator device itself. Shut down the emulator device, then edit the settings for the device in the device manager. Click Show Advanced Settings then ensure your Cameras are [...]

So is there no way to get takePicture to work if your using a physical webcam? 

Why is this issue closed? I still am encountering the `takePicture` bug where it does not return anything.