margelo/react-native-vision-camera #754: ❓ Is it possible to access the JSI runtime inside Frame Processors?

swittk, 2022-01-19:
### Question

I've been bridging a few Machine Learning libraries over to React Native, mostly to use with [react-native-native-video](https://github.com/swittk/react-native-native-video) for video files in University projects, but decided to use VisionCamera for the live camera part.

After implementing my first frame processor for VisionCamera [MLKit Pose Detection](https://github.com/swittk/react-native-mlkit-pose-detection) I noticed that there's no ability to create JSI HostObjects in the callback (there's no `jsi::Runtime` passed anywhere in the callback, nor is the `FrameHostObject.h` header exposed in the VisionCamera's podspec's `s.source_files` field.).

So currently I'm returning the values as plain old NSDictionaries, which would need to be serialized by the runtime again. Personally I believe the ability to use JSI Objects would help increase performance greatly for certain cases (in my case, not needing to serialize the whole detected landmarks (which is a giant NSDictionary) would certainly help).

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-01-19:
Hey! Glad to hear VisionCamera is being used for university projects 😄

 How big is the dictionary? I can't really give you a C++ API here, since that's
1. Complicated as hell to dynamically inject
2. Even more complicated on Android
3. Probably not giving that much performance improvements

swittk, 2022-01-19:
Well I teach at a university, and they're projects of mine... so... I sort of consider them university projects 😂

As for the dictionary.. Well... MLKit has 33 unique strings for the pose points and I'm mapping them to shorter, cross platform keys (see [here](https://github.com/swittk/react-native-mlkit-pose-detection/blob/032b66773a2f7308f6c05b9df491f25fa93dc50b/ios/SKRNMLKitiOSPoseDetector.mm#L16)). Just the keys themselves are pretty long, and inside there's the same key and a few number values for the predicted coordinates too.

I'm not familiar with how your library uses dynamic injection of C++ code, so I can't really say much apart from the fact that my own library, [react-native-native-video](https://github.com/swittk/react-native-native-video), appears to use a different paradigm to class organization (I use common base classes with defined virtual methods, which each native platform then subclass and override).

A benefit to using JSI / expose JSI headers is also lets me reuse the plugins I've written for `react-native-native-video` immediately (they all run on JSI, and just having `jsi::Runtime` should be enough). But.. if it is a major inconvenience then that's OK, I'll just continue adapting my plugins for VisionCamera by using `__has_include()` and writing custom code in Objective-C / Java for the VisionCamera part.

mrousavy, 2022-01-19:
Those keys aren't big at all, I doubt the conversion between NSDictionary -> JS values even takes longer than 1ms.

swittk, 2022-01-19:
Thanks for the reassurance on performance 👍

Just wondering though, I'm toying with the idea of re-using my JSI modules directly in the FrameProcessor callbacks and since I can't access VisionCamera's internal `jsi::Runtime`, do you think I could generate my own runtime, taking inspiration from how it's done in this repository ([iOS](https://github.com/mrousavy/react-native-vision-camera/blob/be5ec69b0292c0a20aaa63e8a9683822d370a056/ios/Frame%20Processor/FrameProcessorRuntimeManager.mm), [Android](https://github.com/mrousavy/react-native-vision-camera/blob/42e791b4bd63234a69b91699d84bf54bf1088db1/android/src/main/cpp/FrameProcessorRuntimeManager.cpp)), then use std::mutex with CallInvoker inside the FrameProcessor callback in order to synchronize the results between the threads? I know it's a bit of a stretch, but if there's nothing wrong with that idea I might give it a go if I have time.

mrousavy, 2023-09-01:
You can now access all JSI things in your worklet runtime

closed 2023-09-01 by mrousavy