margelo/react-native-vision-camera #758: 🐛 Focus on tap

TasigurIshtar, 2022-01-19:
### What were you trying to do?

Hello,

I am trying to simulate focus on tap, but the location where I tapped is newer focussed correctly.

Fresh project - ("react-native": "0.66.4", "react-native-vision-camera": "^2.12.0", ...android device)
Locked portrait orientation,
Camera component has marginTop:60, marginLeft:60 to better check of tap coordinates.
I am using coordinates relative to Camera component rectangle as mentiond here https://mrousavy.com/react-native-vision-camera/docs/guides/focusing/

I made demo project where this problem is ilustrated.
https://github.com/TasigurIshtar/RNVisionCameraTest

Thanks for your time.

### Reproduceable Code

```tsx
const touchII = async (event: NativeTouchEvent) => {
    let point: Point = {
      x: Math.round(event.pageX - camLocation.x),
      y: Math.round(event.pageY - camLocation.y),
    };
    await cameraRef?.current?.focus(point)     
  };

.....
 <Camera
      ref={cameraRef}
      style={{
        marginTop: 60,
        marginLeft: 60,
        width: window.width - 60,
        height: window.height - 60,
      }}
      onLayout={event => {
        const layout = event.nativeEvent.layout;
        setCamLocation({x: layout.x, y: layout.y});
      }}
      device={device}
      isActive={true}
      onTouchEnd={x => device.supportsFocus && touchII(x.nativeEvent)}
    />
```

### What happened instead?

after tapping to the top of the bottle .focus({"x": 53, "y": 393})  -> the bottle doesn't seems to be focused.  _(when I was tapping more on the right site around the grayish mat, occasionaly the top of the bottle was super sharp. )_

[image]

### Device

Samsung Galaxy S10

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-01-19:
Is the behaviour the same on iOS?

savv, 2022-01-20:
Are you sure you are getting the coordinates right? Try:
```
    <TapGestureHandler onHandlerStateChange={({nativeEvent}) => {
      camera.current?.focus({x: nativeEvent.x, y: nativeEvent.y});}}>
  (your camera here)
    </TapGestureHandler>
```

Hajan39, 2022-01-21:
@mrousavy on iOS, the focus is not doing anything in my case
@savv it is doing the same thing

Angelk90, 2022-01-24:
@TasigurIshtar , @Hajan39 : Do both front / back cameras work on android?

Hajan39, 2022-02-08:
Yes, both of them - any progress on this? I think this is crucial functionality ...

mrousavy, 2022-02-08:
Does it also not work in the Example app when you translate the Camera component a bit to the right?

Menardi, 2022-03-31:
I've created a PR to fix this issue (#958). I would appreciate if anyone else could test it out to make sure it fixes the issue for them too. The easiest way to test it is probably to use the patch below (which is the same code as the PR) with [patch-package](https://www.npmjs.com/package/patch-package).

**patches/react-native-vision-camera+2.13.1.patch**
```
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Focus.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Focus.kt
index da68003..c945c0b 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Focus.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Focus.kt
@@ -1,9 +1,9 @@
 package com.mrousavy.camera
 
 import androidx.camera.core.FocusMeteringAction
-import androidx.camera.core.SurfaceOrientedMeteringPointFactory
 import com.facebook.react.bridge.ReadableMap
 import kotlinx.coroutines.guava.await
+import kotlinx.coroutines.withContext
 import java.util.concurrent.TimeUnit
 
 suspend fun CameraView.focus(pointMap: ReadableMap) {
@@ -16,8 +16,11 @@ suspend fun CameraView.focus(pointMap: ReadableMap) {
   val x = pointMap.getDouble("x") * dpi
   val y = pointMap.getDouble("y") * dpi
 
-  val factory = SurfaceOrientedMeteringPointFactory(this.width.toFloat(), this.height.toFloat())
-  val point = factory.createPoint(x.toFloat(), y.toFloat())
+  // Getting the point from the previewView needs to be run on the UI thread
+  val point = withContext(coroutineScope.coroutineContext) {
+    previewView.meteringPointFactory.createPoint(x.toFloat(), y.toFloat());
+  }
+
   val action = FocusMeteringAction.Builder(point, FocusMeteringAction.FLAG_AF or FocusMeteringAction.FLAG_AE)
     .setAutoCancelDuration(5, TimeUnit.SECONDS) // auto-reset after 5 seconds
     .build()
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
index 16eb4dc..507e9fd 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt
@@ -121,7 +121,7 @@ class CameraView(context: Context, private val frameProcessorThread: ExecutorSer
   private val cameraExecutor = Executors.newSingleThreadExecutor()
   internal val takePhotoExecutor = Executors.newSingleThreadExecutor()
   internal val recordVideoExecutor = Executors.newSingleThreadExecutor()
-  private var coroutineScope = CoroutineScope(Dispatchers.Main)
+  internal var coroutineScope = CoroutineScope(Dispatchers.Main)
 
   internal var camera: Camera? = null
   internal var imageCapture: ImageCapture? = null
```

closed 2022-03-31 by mrousavy

brascene, 2023-06-21:
@Hajan39 did you resolved this on iOS? In my case it's also doing nothing, it's throwing error but nothing happens on the screen. I'm using `TapGestureHandler` wrapper to get coordinates, as @savv suggested.

griffinbaker12, 2023-11-26:
For me when I press the screen to focus, the screen just gets more blurry. I am on the iPhone 15.

mrousavy, 2023-11-28:
@griffinbaker12 are you using VisionCamera 3.6.12? For me this works perfectly smooth now ever since I merged https://github.com/mrousavy/react-native-vision-camera/pull/2220 :)

PR #958 (2022-03-31): Fix Android focus not using correct focus point
## What

This PR fixes the `focus` function on Android not focusing on the given point (#758). The issue is that when calculating the focus point with `SurfaceOrientedMeteringPointFactory`, it does not factor in any scaling or cropping that has been done to the camera's view (thanks to [this StackOverflow answer](https://stackoverflow.com/a/60585382) for pointing me in the right direction!). I think it's possible this issue is not obvious unless you're resizing the camera view significantly in some way.

## Changes

Rather than manually calculate [... cut]
commit fe012952 (2022-03-31): fix: Fix Android focus not using correct focus point (#958)

This commit fixes #758. I was having the same issue and looked into it a bit. I found
[this StackOverflow answer](https://stackoverflow.com/a/60585382) which described a
solution to the same problem. Rather than manually calculate the focus point, we can
get the PreviewView to do it for us. This fixes the issue because the PreviewView
factors in any scaling or resizing of the view on the screen, which we weren't doing
before. The only potential issue is that this needs to run on the UI thread
(which is what the `withContext` is doing [... cut]