margelo/react-native-vision-camera #770: ❓ Android resolution ans FPS setting

seolhyeonyang, 2022-01-26:
### Question

I'm a React-Native beginner.
I want to adjust the resolution and FPS on Android.
Currently, when you shoot a video, it comes out at 1920x1080, 30fps.
I want to raise it to 3820 x 2160, 60fps.

```
            <Camera
                ref={camera}
                style={{flex:1}}
                device={phonedata}
                isActive={true}
                video={true}
                audio={true}
                fps={60}
                videoWidth={3840}
                videoHeight={2160}
            />
```
It's my code.

It's information from 'Camera.getAvailableCameraDevices()'.

> {"devices": ["ultra-wide-angle-camera"], "formats": [{"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": true, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 3456, "photoWidth": 4608, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 3456, "videoStabilizationModes": [Array], "videoWidth": 4608}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": true, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 3456, "photoWidth": 4608, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 3456, "videoStabilizationModes": [Array], "videoWidth": 4608}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 2592, "photoWidth": 4608, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 2592, "videoStabilizationModes": [Array], "videoWidth": 4608}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 2184, "photoWidth": 4608, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 2184, "videoStabilizationModes": [Array], "videoWidth": 4608}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 3456, "photoWidth": 3456, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 3456, "videoStabilizationModes": [Array], "videoWidth": 3456}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 2160, "photoWidth": 3840, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 2160, "videoStabilizationModes": [Array], "videoWidth": 3840}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 1080, "photoWidth": 2288, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 1080, "videoStabilizationModes": [Array], "videoWidth": 2288}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 1080, "photoWidth": 1920, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 1080, "videoStabilizationModes": [Array], "videoWidth": 1920}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 912, "photoWidth": 1920, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 912, "videoStabilizationModes": [Array], "videoWidth": 1920}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 1080, "photoWidth": 1440, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 1080, "videoStabilizationModes": [Array], "videoWidth": 1440}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 720, "photoWidth": 1280, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 1280}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 1088, "photoWidth": 1088, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 1088, "videoStabilizationModes": [Array], "videoWidth": 1088}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 720, "photoWidth": 960, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 960}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 480, "photoWidth": 720, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 480, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 480, "photoWidth": 640, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 480, "videoStabilizationModes": [Array], "videoWidth": 640}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 360, "photoWidth": 640, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 360, "videoStabilizationModes": [Array], "videoWidth": 640}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 288, "photoWidth": 352, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 288, "videoStabilizationModes": [Array], "videoWidth": 352}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 240, "photoWidth": 320, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 240, "videoStabilizationModes": [Array], "videoWidth": 320}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": true, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 3456, "photoWidth": 4608, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 3456, "videoStabilizationModes": [Array], "videoWidth": 4608}, {"autoFocusSystem": "none", "colorSpaces": [Array], "fieldOfView": 104.00253234078629, "frameRateRanges": [Array], "isHighestPhotoQualitySupported": false, "maxISO": 800, "maxZoom": 8, "minISO": 40, "photoHeight": 2592, "photoWidth": 4608, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 2592, "videoStabilizationModes": [Array], "videoWidt [... cut ...]
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

normanbrightdev, 2022-01-31:
@seolhyeonyang I encountered this same issue

xulihang, 2022-02-17:
I am also having problems setting up the resolution: https://github.com/mrousavy/react-native-vision-camera/issues/281#issuecomment-1042649302

closed 2023-09-01 by mrousavy