margelo/react-native-vision-camera #777: Would you be interested in supporting react-native-windows❓

creambyemute, 2022-01-28:
### Question

Would you be interested in supporting react-native-windows with this library?

RN-Camera has been deprecated so I'm looking for a library to migrate to in the future.

I'd volunteer to do the implementation, but probably not in the next few weeks as we're currently focused on some other tasks before I'd get around to app-stuff again.

### VisionCamera Version

None yet

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-01-31:
Hi Thomas!

Yes, I have been in touch with Microsoft (@kelset) before talking about this. My opinion back then was, and still is:

I'd like to keep a "lean core" for VisionCamera, that's why the foundation for Frame Processors exists, but not Barcode reader. The community can create a Barcode reader themselves, and even choose between implementations (MLKit, Vision, ...). The same principle applies to platforms. VisionCamera can be implemented on react-native-windows, and this is easily possible by creating a separate repository (e.g. react-native-vision-camera-windows) that just implements the bindings (`NativeModules.CameraView`) for Windows. 
This allows for an implementation without touching react-native-vision-camera, without having the user download a larger npm package, without having the user wondering why there are so many updates but nothing changes (because everything might be windows related and he only uses Android and iOS), and so on.

I would love to help out (especially with the Frame Processors feature, I looked into it and it seemed fairly easy on Windows), and obviously answer any questions regarding the implementation if you're down to do it!

Also, I don't think any huge breaking changes are coming up, and if they are, I will obviously coordinate this with the maintainers of react-native-vision-camera-windows.

shaikhaffan, 2022-02-10:
Thanks you @mrousavy would you mind letting us know the timeline for windows package? I would love to try it on windows.

mrousavy, 2022-02-11:
Hi @shaikhaffan, as said in my previous comment, I am not working on adding Windows support. I am willing to lend a hand on some complex tasks (e.g. Frame Processors + JSI) and can answer implementation questions, but I am not implementing it myself, especially not in my free time. I don't work on Windows.

closed 2023-09-30 by mrousavy