margelo/react-native-vision-camera #779: 🐛 Camera: Native Module 'CameraView' was null! Did you run pod install?

saurabhkanswal, 2022-01-31:
### What were you trying to do?

trying to set up a vision camera with the project.
Have tried all things in troubleshoot guide.

### What happened instead?

I am getting an error message:
`Camera: Native Module 'CameraView' was null! Did you run pod install?`

I have also done pod install.
I don't have any idea, why it is not working...

android:
```
        buildToolsVersion = "30.0.0"
        minSdkVersion = 21
        compileSdkVersion = 31
        targetSdkVersion = 31
        supportLibVersion = "27.0.0"
        kotlinVersion = '1.5.0'
        ndkVersion = "21.4.7075529"
```

### Device

Android  realme Narzo 20a

### VisionCamera Version

^2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-01-31:
Did you rebuild the android app?

saurabhkanswal, 2022-01-31:
yeah i did that, still issue is same.

mrousavy, 2022-01-31:
Which RN version

saurabhkanswal, 2022-01-31:
0.63.2

ArthurDias01, 2022-02-01:
facing same issue here,

also using expo (SDK 44.0)
RN 0.64.3

paul-castro, 2022-02-04:
any update on this?

ArthurDias01, 2022-02-04:
> any update on this?

I surrendered to Update macOS to Monterey and then updated Xcode to 13.0, after that everything went fine....

matpandino, 2022-02-11:
same issue here

madmeatballs, 2022-03-24:
same issue, is there a solution?

    expo: ~44.0.0
    react-native: 0.64.3

zhenguet, 2022-03-30:
have any update for this?

b-alidra, 2022-04-23:
Hi, I have the same issue and I'm not using expo...
RN 0.65.1

b-alidra, 2022-04-23:
> I have the same issue and I'm not using expo...
> 
> ```
> RN 0.65.1
> VC 2.13.0
> ```

I've fixed the problem on my side: I was just missing `packages.add(new CameraPackage());` in my `MainApplication.java`...

muyi0327, 2022-10-14:
I have the same issue，because i forget to add 'react-native-reanimated/plugin' to babel plugins. like this:

[https://docs.swmansion.com/react-native-reanimated/docs/fundamentals/installation/](react-native-reanimated)

I've fixed the problem!

fizefeko, 2022-11-02:
> > I have the same issue and I'm not using expo...
> > ```
> > RN 0.65.1
> > VC 2.13.0
> > ```
> 
> I've fixed the problem on my side: I was just missing `packages.add(new CameraPackage());` in my `MainApplication.java`...

Can you please tell me from which package did you import the CameraPackage class :)

gkartalis, 2023-02-03:
I was getting the same error but only when running tests, adding this to our `setupJest` file fixed the issue:

```js
jest.mock("react-native-vision-camera", () => {})
```

closed 2023-08-18 by mrousavy