margelo/react-native-vision-camera #786: ❓ Android build failed - it's not clear what exactly is the issue

vvusts, 2022-01-31:
### Question

I have older RN project that is on **RN: 0.63.4**.
I am on a way to migrate from RNCamera to this library.
I installed **"react-native-vision-camera": "^2.12.0"**.
```
buildToolsVersion = "29.0.2"
minSdkVersion = 21
compileSdkVersion = 31
targetSdkVersion = 30
```

But I am getting android build errors like:
```
> Task :app:processDebugResources FAILED
w: Detected multiple Kotlin daemon sessions at build/kotlin/sessions

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
...
Execution failed for task ':app:processDebugResources'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.Workers$ActionFacade
   > Android resource linking failed
...
/.gradle/caches/transforms-2/files-2.1/29678a57a31f9429c3228f99a5a6aa7c/camera-extensions-1.0.0-alpha32/AndroidManifest.xml:24:5-28:15: AAPT: error: unexpected element <queries> found in <manifest>.
```

I don't know what to try anymore. Is there some issue with this RN version?
I don;t use expo.

### What I tried

Clear build folder.
./ gradelw clean > build successfull.
Rebuilding project multiple times on real device.

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

jonotrybe, 2022-01-31:
What is the version of your android plugin?
https://stackoverflow.com/questions/62969917/how-to-fix-unexpected-element-queries-found-in-manifest-error

based on this I tink you may need to do some android upgrades. This is not specifically an issue with this lib but that the CameraX tool is updated to support Android 11

jonotrybe, 2022-01-31:
I am current looking inside my project, I just upgraded.

in this file: `android/build.gradle`

I have this entry
` classpath("com.android.tools.build:gradle:4.1.2")`

vvusts, 2022-02-01:
Somehow I missed to update that value build passed, thanks :)

closed 2022-02-01 by vvusts

NithinSK04, 2022-02-01:
@vvusts how did you fix it?. can you elaborate ?