margelo/react-native-vision-camera #787: Android crashes every time, when I change global state or when I navigate away from navigator with camera screen🐛

Deodes, 2022-01-31:
### What were you trying to do?

I would like to have a possibility to scan qr codes. I've installed react-native-vision-camera and frame processor plugin for this. Everything works great, but when I try to navigate away from screen(or to make camera unmount) with camera or changing state in redux-store, I'm getting crash.

I've already tried to patch the lib with `patch-package`, but it not works well with this package..

### Reproduceable Code

```tsx
<Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={isFocused}
        frameProcessor={frameProcessor}
        frameProcessorFps={1}
        torch={torch}
        zoom={zoom}
        audio={false}
        video={false}
/>
```

### What happened instead?

[image]

### Device

Samsung A-22

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

saurabhkanswal, 2022-01-31:
@Deodes which RN version you are using ?

Deodes, 2022-02-01:
@saurabhkanswal 0.66.2

sutuioncode, 2022-02-01:
This is Workaround.  You have to render the camera only when the camera page is on display. if you are using `react-navigation` you can use `useIsFocused()` hook.
```
const isScreenActive = useIsFocused();
return (<>
{isScreenActive ?
       (<Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={isFocused}
        frameProcessor={frameProcessor}
        frameProcessorFps={1}
        torch={torch}
        zoom={zoom}
        audio={false}
        video={false}/>)
: <View>}
</>)
```

Deodes, 2022-02-02:
@sutuioncode indeed your code works, thank you very much! But I strongly ask the creators to pay attention to this behavior, in my opinion there should be no such unpredictable bugs. Everything else works great, thanks for creating such a great lib!

closed 2022-02-02 by Deodes

mrousavy, 2022-02-02:
> thanks for creating such a great lib

Thank you for the kind words!

> in my opinion there should be no such unpredictable bugs

This is pretty obvious. I am trying my best at fixing bugs and creating new awesome features, but I fully work on this library in my free time. I already attempted to fix the bug you're experiencing 3 times, never found a good solution and ran out of time/patience because I am running a company and don't have a lot of free time.
If you want to show your support, you can consider sponsoring me on GitHub, which helps me get to issues faster.