margelo/react-native-vision-camera #788: ✨ Output Bitrate Modification

xHeinrich, 2022-01-31:
### What feature or enhancement are you suggesting?

In react-native-camera we had the ability to change the video bit-rate. This is particularly useful if you're using the component to send the video up to a server on a poor connection. 

See attached documentation for how it worked there(videoBitrate prop):
https://react-native-camera.github.io/react-native-camera/docs/api#method-type-1

Places in their code this was able to be modified:
https://github.com/react-native-camera/react-native-camera/blob/v4.2.1/ios/RN/RNCamera.m#L1207-L1215
https://github.com/react-native-camera/react-native-camera/blob/v4.2.1/android/src/main/java/com/google/android/cameraview/Camera2.java#L1422

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

You could do post processing on the device using FFMpeg but that seems like a poor idea if you can modify the bitrate during recording.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2022-02-02:
Hi! Yes, this should probably be possible very easily on iOS since I am using my own custom file writer, as opposed to RNCamera which just uses the default iOS file writer. 

Not sure how I can achieve it on Android for now.

siddharth-kt, 2022-03-02:
Hi @mrousavy,
Is there any way for now to achieve this(bitrate modification) on android ?

dylan-westbury, 2022-05-31:
Could this be why "medium" preset still has large videos for android and not iOS?

For a 1 minute video, its around 12mb for ios, 140mb for android

dylan-westbury, 2022-06-01:
Unfortunately we had to revert back to RNCamera because we can't get the android video size down, hopefully someone can help out with setting the bit rate for Android in future 🙂

jpike88, 2022-08-03:
[image]

https://github.com/react-native-camera/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java

It looks like the API isn't even ready yet, @mrousavy do you think you jumped a gun a little on that? Bitrate setting is extremely important, and I don't want a customer waiting for their video to be reencoded all over again because of a shortcoming of the new API.

jpike88, 2022-08-03:
Wait... I may have found something

https://developer.android.com/reference/kotlin/androidx/camera/core/impl/VideoCaptureConfig

jpike88, 2022-08-03:
I've almost lost my mind reading the new androidX documentation, apparently it was possible to manipulate bitrate using VideoCaptureConfig and VideoCapture.Builder, but they seem to have blocked that and only allow these STUPID preset QualitySelector things.

I can't describe how much I hate the Android development team, I can't believe that they would actually make it impossible to touch bitrate settings when recording a video. God I'm so angry

jpike88, 2022-08-03:
Ok from what I understand, the AndroidX Camera library is limited in what it can do. The Camera2 API is NOT deprecated, and was implemented here in react native camera.

I'm going to fork and move to that API for finer control including bitrate, and the 'medium' and 'high' presets on android are creating huge video files.

https://github.com/react-native-camera/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera2.java

saurabhkanswal, 2022-08-13:
@jpike88 do share your library here in this thread.
we are also facing the same issue, the video size is too large and we have to compress it before uploading.
earlier we had used **react-native-compressor** but it increased  our apk size a lot.

Also you can use https://github.com/Tourenathan-G5organisation/SiliCompressor, which use mediaCodec api to compress the video. But the only problem with this library is that it take a lots of time to compress.
eg, for 2 min video it take 4 min compression time and it changes from device to device.

AlkanV, 2022-11-23:
hello, any news on this one? we need to reduce android file size during recording, because all other methods requires post processing time which could be huge burden for the app. 

any help would be appreciated, thanks.

@mrousavy

techied, 2023-09-11:
> Sure! If it's that important for you/your business/your clients, reach out to me over email or Twitter and we can work a way out so I can work on that! :)

Hi! We are using VisionCamera V3 and would like to get the file size down so the video can be uploaded quickly over cellular. We're recording in H265 at the lowest resolution iOS allows, but the file size is still too large for weaker connections. Bitrate control is something we are very interested in. 

I saw #1555 , is this something you're still working on? What would it take to get this feature?

mrousavy, 2023-09-13:
@techied sure, I can implement this as consultancy service. DM me on Twitter (@mrousavy) or send me an email to discuss details :)

mrousavy, 2023-09-29:
See #1882 ;)

closed 2023-09-29 by mrousavy