margelo/react-native-vision-camera #789: 🐛 Camera not initialized after access to camera is granted.

arthwood, 2022-02-01:
### What were you trying to do?

I wanted to be able to take picture after visiting screen with `Camera` component for the first time.

### Reproduceable Code

```tsx
const PictureScreen: FC<Props> = () => {
  const isFocused = useIsFocused()
  const cameraRef = useRef<Nullable<Camera>>(null)
  const { current: cameraRefCurrent } = cameraRef
  const { back: cameraDevice } = useCameraDevices()
  const onCameraInitialized = useCallback(() => console.log('camera initialized'), [])
  const [cameraLayout, setCameraLayout] = useState<LayoutRectangle>()
  const onCameraLayout = useCallback<Dispatch<LayoutChangeEvent>>(
    ({ nativeEvent: { layout } }) => setCameraLayout(layout),
    [setCameraLayout],
  )
  const [cameraPermissionStatus, setCameraPermissionStatus] = useState<CameraPermissionStatus>()
  const isCameraAuthorized = cameraPermissionStatus === 'authorized'

  useEffect(() => {
    Camera.getCameraPermissionStatus().then(setCameraPermissionStatus)
  }, [setCameraPermissionStatus])

  useEffect(() => {
    if (!isCameraAuthorized) Camera.requestCameraPermission().then(setCameraPermissionStatus)
  }, [isCameraAuthorized, setCameraPermissionStatus])

  return 
    <View>
      {cameraDevice ? (
        <Camera
          audio={false}
          device={cameraDevice}
          isActive={isFocused}
          onInitialized={onCameraInitialized}
          onLayout={onCameraLayout}
          photo
          ref={cameraRef}
          style={[styles.camera, isCameraAuthorized ? null : styles.cameraNotAuthorized]}
          video
        />
      ) : (
        <CameraLoader style={styles.camera} />
      )}
    </View>
}

const CameraLoader = View

const styles = StyleSheet.create({
  camera: {
    flex: 1,
    justifyContent: 'flex-end',
    paddingBottom: 10,
  },
  cameraNotAuthorized: {},
}
```

### What happened instead?

On first visit a permission modal is presented (`cameraPermissionStatus` outputs `'denied'`). After access to camera is granted (`cameraPermissionStatus` outputs `'authorized'`) Camera is **not** automatically activated, also `onInitialized` is never being called.
The only way to activate Camera is to reload the app (in development mode or by killing and reopening in release mode) - `onInitialized` is being called then.

Note I applied fake `cameraNotAuthorized` style on `isCameraAuthorized` flag change, just to check if it would force re-render the Camera element - however this doesn't help.

**react**: 17.0.2
**react-native**: 0.66.1

### Device

Samsung Galaxy A50

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

vozaldi, 2022-02-01:
After trying to figure out the issue, I solved this by hiding the `<Camera />` before the permission is granted.

```
return (
  <View>
    {'authorized' !== cameraPermissionStatus ? (
      <Text>Please allow the app to access your camera.</Text>
    ) : (
      cameraDevice ? (
        <Camera
          audio={false}
          device={cameraDevice}
          isActive={isFocused}
          onInitialized={onCameraInitialized}
          onLayout={onCameraLayout}
          photo
          ref={cameraRef}
          style={[styles.camera, isCameraAuthorized ? null : styles.cameraNotAuthorized]}
          video
        />
      ) : (
        <CameraLoader style={styles.camera} />
      )
    )}
  </View>
)
```

mrousavy, 2022-02-02:
Yes, you need to hide the Camera until your app is authorized to use the Camera.

closed 2022-02-02 by mrousavy

arthwood, 2022-02-02:
@vozaldi @mrousavy Thanks, I will do that!

NLanese, 2022-11-15:
This solution did not work for me, after experiencing the same issue. I already have it so that before access is granted, no camera appears. I just get the error as soon as I try to start recording

Hongdeshuai, 2023-02-20:
Yeah, this appears right after I takePhoto

dsipahioglu, 2023-09-03:
I have the same issue when I set video={true} in V3. Photo works fine though.

mrousavy, 2023-09-04:
@dsipahioglu can you create a new issue with accurate reproduction steps?

Joanna-Philips, 2023-09-27:
@dsipahioglu Would you mind sharing your code for capturing the photo. I cannot get it to work even for that.

dsipahioglu, 2023-09-27:
> @dsipahioglu Would you mind sharing your code for capturing the photo. I cannot get it to work even for that.

I've created an issue and there I've shared my code. I hope it helps! https://github.com/mrousavy/react-native-vision-camera/issues/1769