margelo/react-native-vision-camera #790: Build failed  Error':react-native-fast-image:compileDebugJavaWithJavac'.❓

NithinSK04, 2022-02-01:
### Question

initially i ran into a error
"The minCompileSdk (31) specified in a
     dependency's AAR metadata (META-INF/com/android/build/gradle/aar-metadata.properties)
     is greater than this module's compileSdkVersion (android-30).
     Dependency: androidx.camera:camera-view:1.0.0-alpha32."
 
and after changing my minCompileSdk to 31 :

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-fast-image:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

### VisionCamera Version

"react-native-vision-camera": "^2.12.0",

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-02-01:
Your second error is not complete. You didn't include an actual error message.

NithinSK04, 2022-02-01:
@mrousavy my bad.! I've updated it now. please check

NithinSK04, 2022-02-01:
@jonotrybe Oh.. I'll look into it ,Thank you

jonotrybe, 2022-02-01:
> @jonotrybe Oh.. I'll look into it ,Thank you

you will know if the stack trace on the failure (above what you pasted) shows that there was an error in the Java Compiler.

NithinSK04, 2022-02-01:
> one issue i find is, if you are using adoptjdk then this can happen. In Android Studio you can set to use the embedded JDK with the studio. once I did that and also fixed my appcenter build to use it then my issues were resolved

How do i set it to embedded JDK?

jonotrybe, 2022-02-01:
I'd look here: 
https://stackoverflow.com/a/30631386

mrousavy, 2022-02-02:
There is still no actual error message here, looks like you have outdated tooling (Java, Android, ...)

closed 2022-02-02 by mrousavy