margelo/react-native-vision-camera #793: 🐛 Orientation is always 0

vvusts, 2022-02-02:
### What were you trying to do?

Tested on android 8 device (huawei p10 Lite).
I tried to take photos in portrait or landscape but in response object 
`metadata.Orientation` is always zero.
Reason for this is if user take photo in landscape we check orientation and based on that set different UI elements.
Or we need to make sure that photo is taken in desired orientation.

### Reproduceable Code

```tsx
<Camera
        ref={camera}
        style={{position: 'absolute', top: 0, left: 0, bottom: 0, right: 0}}
        device={device}
        isActive={isCameraActive}
        photo={true}
/>
```

### What happened instead?

metadata.Orientatio is always 0

### Device

huawei p10 lite

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2022-02-02:
Does this work on an iPhone?

vvusts, 2022-02-02:
I will test and inform you as soon as possible.

vvusts, 2022-02-02:
Probably will not test soon I am having some:
`/Users/.../.../.../ios/Pods/Headers/Public/ReactCommon/ReactCommon/TurboModuleUtils.h:13:10: 'folly/Optional.h' file not found`
Issue and can't build for ios.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy