margelo/react-native-vision-camera #796: The given view (ID 1549) was not found in the view manager❓

NithinSK04, 2022-02-02:
### Question

Whenever i try to capture a photo i get this error

```
[system/view-not-found: [system/view-not-found] The given view (ID 1749) was not found in the view manager.]

```

### What I tried

 i tried 
```

const CapureImage = async () => {
    if (camera) {
      try {
        const photo = await camera.current.takePhoto({
          quality: 85,
          skipMetadata: true,
        });
        console.log(photo);
      } catch (error) {
        console.log(error);
      }
    }
  };

```

### VisionCamera Version

"react-native-vision-camera": "^2.12.0",

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-02-03:
Hi! This is a duplicate issue, I remember another one being exactly the same about not working in a Modal. You can use a separate Screen and present it as a Modal to achieve the same effect

closed 2022-02-03 by mrousavy

iRootPro, 2022-02-17:
Good Day!
A get same problem, but i unused modal
<Camera/> wrapper in View elements only, but i click on takePhoto on android i get [system/view-not-found] The given view (ID 3677) was not found in the view manager.

on IOS work is fine, but android failed (