margelo/react-native-vision-camera #799: ✨ White balance should be controllable

thomaschaaf, 2022-02-04:
### What feature or enhancement are you suggesting?

It would be very good if we could set the white balance of the camera. Optimally even setting the temperature ourselves.

RNCamera had whiteBalance as a feature (https://react-native-camera.github.io/react-native-camera/docs/rncamera#whitebalance)

https://developer.apple.com/documentation/avfoundation/avcapturedevice/whitebalancemode

https://developer.android.com/reference/android/hardware/camera2/CaptureRequest#CONTROL_AWB_MODE

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

Currently the images just look bad in artificial light for example.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2022-02-04:
Hi!

Thanks for the well written feature-request, I can see how `whiteBalance` should be an important prop.

I am using CameraX instead of Camera2, but I doubt that CameraX is lacking this feature, so I can maybe take a look soon :)

paulrostorp, 2022-02-09:
I could look into the ios implementation

mrousavy, 2023-09-30:
hey - now in V3 this is much easier to implement! I think I'll throw a bounty on this

kinghoddy, 2025-06-21:
Any update on this ?