margelo/react-native-vision-camera #800: Issues with my custom frameProcessor: textRecognition Empty on iOS❓

digitalbore, 2022-02-04:
### Question

I successfully created my custom frameProcessor to perform TextRecognition.
The frameProcessor is correctly linked with React-Native.

But inside the Objective C implementation, the method to recognize text always return empty values, even if the camera is framing texts:

```
static inline id scanOCR(Frame* frame, NSArray* args) {
  MLKTextRecognizer *textRecognizer = [MLKTextRecognizer textRecognizer];
  MLKVisionImage *image = [[MLKVisionImage alloc] initWithBuffer:frame.buffer];
  image.orientation = frame.orientation;
  
  NSError *error;
  MLKText *result = [textRecognizer resultsInImage:image error:&error];
  if (error != nil || result == nil) {
    NSLog(@"%@", error); // <- This is NEVER called
  } else {
    NSLog(@"text: %@", result.text); // <- this is always empty
    NSLog(@"blocks lenght: %lu", (unsigned long)result.blocks.count); // <- this is always 0
  }
  return @{@"text": result.text};  
}
```

All the code is correctly built and compiled. I have no error messages but I cannot recognize any texts.

What is happening?

### VisionCamera Version

^2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-02-04:
Hi! This seems to be an issue with the MLKit Text Recognizer, not with VisionCamera. 

Maybe MLKit does not support 10-bit HDR Frames yet, and your Camera is using 10-bit HDR Frame Format? Check out [`CameraDeviceFormat.pixelFormat`](https://mrousavy.com/react-native-vision-camera/docs/api/interfaces/CameraDeviceFormat#pixelformat), you want that to be `420f` or `420v` instead of `x420`.

closed 2022-02-04 by mrousavy

digitalbore, 2022-02-04:
@mrousavy that's a good suggestion, but I just tried with various **formats** and **preset** and text results is always empty. Yes it is related with MLKit, more than VisionCamera.

mrousavy, 2022-02-04:
Hmm, that sucks, sorry I'd have to debug this myself. If the Frame appears to be correct, which you can verify by writing it to a file:

```swift
let imageBuffer = CMSampleBufferGetImageBuffer(sampleBuffer)!
let ciimage = CIImage(cvPixelBuffer: imageBuffer)   

let context = CIContext(options: nil)
let cgImage = context.createCGImage(ciimage, from: ciimage.extent)!
let image = UIImage(cgImage: cgImage)

let paths = NSSearchPathForDirectoriesInDomains(.documentDirectory, .userDomainMask, true)
let filePath = "\(paths[0])/MyImageName.png"
UIImagePNGRepresentation(image)?.writeToFile(filePath, atomically: true)
```

And if that looks correct, then it must be MLKit's fault.

digitalbore, 2022-02-06:
Hi @mrousavy , I was able to solve the problem by updating the iOS version of my iPhone 8 from 13.6 to 15.3.

Do you know why that could happen?

Do you know if MLKIT compatible with older iOS versions?

mrousavy, 2022-02-06:
Hi, I don't know.

rzhbishal, 2023-07-11:
hello @mrousavy  @digitalbore  I have the same error. I am using vision camera ocr and react-native-vision-camera. I upgraded the mlkit inside vision-camera-ocr from 2.2.0 to 3.2.0 and i am getting problem like everything builds ok it's just text is not being displayed. 

`2023-07-11 16:41:33.700470+0900 VisionCameraOcrExample[5211:1423971] [native] VisionCamera.invokeOnFrameProcessorPerformanceSuggestionAvailable(currentFps:suggestedFps:): Frame Processor Performance Suggestion available!`