margelo/react-native-vision-camera #801: "Mapping" between frame coordinates and actual coordinates in Camera view on screen❓

Feel2X, 2022-02-05:
### Question

Hello,

first of all, thank you for this great work!!

I face difficulties when trying to map object detection bounding box coordinates from the actual video frame to the frames as they are displayed on the phone screen. 
I keep the Camera view in a smaller container filling only the top side of the screen. Therefore, parts of the frames are not visible and they appear to be scaled down. Is there a clever way of mapping the bounding box coordinates to the corresponding coordinates in my scaled-down Camera view?

Thank you and best regards,
Felix

### What I tried

I tried to work with relative bounding box coordinates, but this does not consider the "cropping" and scaling operations on the frame when it is displayed in a small container.

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-02-08:
Hey Felix!

Thanks!

Well I faced this problem a while ago too. In my case, I just had the selected `format` (one of `device.formats`) in memory, pass that to the `<Camera>` component, then measure the `Camera` component on-screen and calculate the point using the information I have (I know Camera X and Y position, I know Camera View width and height, and I know actual Camera format dimension width and height). This means it's just a little bit of basic Maths.

I am waiting on Fabric to be released, so I can build a simple `camera.toScreenCoordinates(rect)` function on some bounding rects that will make this very easy. :)

Feel2X, 2022-02-08:
Hey Marc!

thanks for your fast reply!

I also settled with this solution now. This is working perfectly if I use a camera format with height == width, and make sure that the camera view on screen has the 1:1 aspect ratio as well.
However, if I choose a format with height != width, and apply the corresponding aspect ratio to the camera view, there still is an offset for my bounding box. I guess there is some kind of cropping operation performed on one of the dimensions so I will have to dig deeper into understanding this.

Again, thank you very much! I am really looking forward to something like `camera.toScreenCoordinates(rect)` in the future!

closed 2022-02-08 by Feel2X

xulihang, 2022-02-22:
I use react-native-svg to draw overlays. It works okay with a 16:9 resolution. I've written about this in a blog: https://www.dynamsoft.com/codepool/react-native-qr-code-scanner-vision-camera.html