margelo/react-native-vision-camera #810: Error: Exception in HostFunction: com.mrousavy.camera.ViewNotFoundError🐛

fabio-C, 2022-02-08:
### What were you trying to do?

My Android app is in production and is using **react-native-vision-camera** with a custom **frameProcessor**.

I am receiving several crashes error notification from Google Play Console (100 crashes in the last 72hours).

I attach one:
[image]

Do you know why this is happening?
Can be a problem with my custom frameProcessor? 
And what can I do to debug it? My app never crashed during development.

Thanks for your help

### What happened instead?

I don't known.

### Device

Android 11, Android 10, Android 12

### VisionCamera Version

^2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

gelfer, 2022-02-09:
#626 

you should check this out. Similar issue and still under investigation.

mrousavy, 2022-02-09:
hmm, yes this will likely be solved with Fabric

digitalbore, 2022-02-17:
Hi @mrousavy 

I temporarily solved the bug commenting line 417 on Camera.tsx

```
  /** @internal */
  componentWillUnmount(): void {
    if (!this.isNativeViewMounted) return;
    if (this.lastFrameProcessor != null || this.props.frameProcessor != null) {
      // this.unsetFrameProcessor(); // <-- this one
      this.lastFrameProcessor = undefined;
    }
  }
```
Does this seem like a valid temporary solution or could something worse happen? 
When do you think the bug will be fixed?

mrousavy, 2022-02-17:
actually I think this can be removed, it will delete the memory on the native side anyways 🤔

adamlbailey, 2022-02-23:
Also getting this same error intermittently in production: 
```
Error: Exception in HostFunction: com.mrousavy.camera.ViewNotFoundError: [system/view-not-found] The given view (ID 5647) was not found in the view manager., js engine: hermes, stack:
setFrameProcessor@1:5172069
onViewReady@1:5172162
invokeGuardedCallbackImpl@1:109371
invokeGuardedCallback@1:109428
invokeGuardedCallbackAndCatchFirstError@1:109460
executeDispatch@1:109592
executeDispatchesAndReleaseTopLevel@1:112988
forEachAccumulated@1:111074
anonymous@1:113345
batchedUpdatesImpl@1:168436
batchedUpdates@1:112905
_receiveRootNodeIDEvent@1:113182
receiveEvent@1:162582
__callFunction@1:179852
anonymous@1:178472
__guard@1:179418
callFunctionReturnFlushedQueue@1:178430
```

tarikko, 2022-02-28:
```
  /** @internal */
  componentWillUnmount(): void {
    if (!this.isNativeViewMounted) return;
    if (this.lastFrameProcessor != null || this.props.frameProcessor != null) {
      // this.unsetFrameProcessor(); // <-- this one
      this.lastFrameProcessor = undefined;
    }
  }
```
Removing that line fixed it for me ! I wish that the next version will permanently fix this annoying issue.

mrousavy, 2022-02-28:
@adamlbailey instead of putting a bug bounty on this specific issue, consider [sponsoring me](https://github.com/sponsors/mrousavy) for all the other parts of the lib that work perfectly fine :)
Also, I already fixed the issue but forgot to release a new version to npm. Doing that now (cc @tarikko)

adamlbailey, 2022-02-28:
Sponsored with pleasure. Thanks for the good work @mrousavy!

andrepaulo-bit, 2022-03-01:
Hello! 
Was this issue fixed in 2.12.1? 
I'm using a Camera in a modal from: https://www.npmjs.com/package/react-native-modal and https://github.com/rodgomesc/vision-camera-code-scanner but I'm still getting the same error.

adamlbailey, 2022-03-10:
Also still getting the same error after upgrading to 2.12.1.

PABourdais, 2022-03-10:
> It was happened to me when i put the component as a children inside of React Native's Modal component That occurs only on android btw
> 
> To temporary solution, remove modal view and use view component

same for me on 2.12.2

gary-cohen, 2022-03-15:
Same here. I'm not sure I can remove it from the modal though.

cjanietz, 2022-03-23:
Currently the current activity is used to recover the view instance from the view id tag.
However to deal with different view hierarchies as introduced by View Groups elsewhere in the tree it is probably better to utilize the React ViewManager for resolving the view instance.

```kotlin
val view = UIManagerHelper.getUIManager(mContext.get(), viewId).resolveView(viewId)
```
should fix this here:
https://github.com/mrousavy/react-native-vision-camera/blob/be5ec69b0292c0a20aaa63e8a9683822d370a056/android/src/main/java/com/mrousavy/camera/frameprocessor/FrameProcessorRuntimeManager.kt#L63

mrousavy, 2022-03-23:
Does that fix the issue? If yes, shoot a PR and I'll test & merge!

cjanietz, 2022-03-23:
> Does that fix the issue? If yes, shoot a PR and I'll test & merge!

Yes it does, for me camera now runs perfectly in a modal

closed 2022-03-28 by mrousavy

adembacajmemmo, 2022-08-29:
@mrousavy I think would be good to specify in README that users need to upgrade React Native version to 0.66.0 or above in order to use react-native-vision-camera 2.13.1 and above.

I just faced this issue, and after upgrading to 2.13.1, I couldn't build Android anymore. After you used UManager, there is resolveView which has been introduced in react-native 0.66.0 and above, so after upgrading react-native, it's working good now.

3imed-jaberi, 2024-07-03:
If anyone passes by here, you can fix this issue by unmounting the camera component/screen once you don't need it ...

PR #928 (2022-03-28): Fix issue with resolving camera view in Frame Processor
## What

When a modal is used the Camera view is not in the hierarchy of the current activity.
Therefore we utilize the global React Native View registry to resolve the view instance.

## Changes

Changed way how Camera view is resolved

## Tested on

Android 12 

## Related issues

Fixes #810