margelo/react-native-vision-camera #815: ❓ Frame Processor threw an error: Value is undefined, expected an Object

meridPL, 2022-02-09:
### Question

Hi,

When I add to the Component frameProcessor as props when I get that error from title.

In index.tsx I have add `import 'react-native-reanimated';`

This it Camera component:
`
      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
        frameProcessor={frameProcessor}
      />
`

and function frameProcessor:

`
const frameProcessor = useFrameProcessor((frame: Frame) => {
    'worklet'
    console.log("Not Hotdog.")
  }, [])
`

Maybe I somewhere get mistake but I not any idea where :)

### VisionCamera Version

2.12.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2022-02-09:
Hey! Maybe you need to add the import to the file where you use `useFrameProcessor` instead of just index.tsx?

mrousavy, 2022-02-09:
Did you also follow [the installation guide](https://docs.swmansion.com/react-native-reanimated/docs/fundamentals/installation)?

meridPL, 2022-02-09:
From intruction reanimate I done:

`module.exports = {
  presets: ['module:metro-react-native-babel-preset'],
  plugins: ['react-native-reanimated/plugin'],
};`

`project.ext.react = [
  enableHermes: true  // <- here | clean and rebuild if changing
]`

`import com.facebook.react.bridge.JSIModulePackage; // <- add 
import com.swmansion.reanimated.ReanimatedJSIModulePackage; // <- add
`
and

`  @Override
      protected JSIModulePackage getJSIModulePackage() {
        return new ReanimatedJSIModulePackage(); // <- add
      }
`

I don't use progroud.

I added also import to App.jsx but it not resolved my problem.

meridPL, 2022-02-09:
I am cleaning my android project for each time

mrousavy, 2022-02-09:
Hmm, did you start with clean metro cache as well? `yarn start --reset-cache`

meridPL, 2022-02-09:
Ok thanks, It started work, maybe reset metro was needed.

mrousavy, 2022-02-09:
Great! Let me know how you like the library :)

closed 2022-02-09 by mrousavy

meridPL, 2022-02-09:
I have yet one question.
In the older version RNCamera was prop onTextRecognized. Then I could used like it `onTextRecognized={({textBlocks})=>{console.log(textBlocks)}}`.

How I can do something similar in your library?
I saw useFrameProcessor, but it is very complicated for me.

dungnguyentri3110, 2022-10-07:
have any solution?. when i build release apk faced with this. Work well on mode debug.